Part-DB/Part-DB-server #787: Farnell API key, disabled information provider

dominykasLi, 2024-12-02:
Hello,

I did the setup for partdb and apache2 server (not through docker) a few days ago. Everything works, except I cant get the information providers to work. 
[image]

I added to /var/www/partdb/config/parameters.yaml these two lines:
 env(PROVIDER_ELEMENT14_KEY): 'zgwfnsaxvhzuwftfxhqxxxxx'
 env(PROVIDER_ELEMENT14_STORE_ID): 'lt.farnell.com'

then did:
sudo -u www-data php bin/console cache:clear

It works and the I refresh the partdb page and nothing changes, it is still disabled.

**Desktop (please complete the following information):**
 - OS: win10
 - Browser chrome

frank-f, 2024-12-07:
Don't change that file, it will be overwritten on each update. Modify docroot/.env.local instead.

I'm not sure if that will solve your issue, but it will save you some hassle during future updates.

dominykasLi, 2024-12-12:
> Don't change that file, it will be overwritten on each update. Modify docroot/.env.local instead.
> 
> I'm not sure if that will solve your issue, but it will save you some hassle during future updates.

That did solve the problem. I just added the API keys to the /var/www/partdb/.env.local and everything worked. Thank you.

closed 2024-12-12 by dominykasLi