Part-DB/Part-DB-server #789: Export of parts

jamel-86, 2024-12-06:
Hi,

Amazing tool!

Can someone explain to me how to export all parts in my library?

I was trying to follow the instructions in the documentation, and here is what I did:

1. I select the top level of all my categories.
2. Then I select all parts in the table.
3. After this, it says there is supposed to be an appearing menu, however, I'm not able to find this menu? 

Maybe I'm on the wrong page, I also tried to same by going to edit, then categories. Here I can find the export option, but even if select include children, only the categories will be exported ....

[image]

serdenberger, 2025-01-01:
Hi,
i like the tool. Its amazing.

I have the same issue. I can export categories, manufacturers, ...
But i can't find a export all parts. 
I use the docker version (v1.14.5)
Try with user and admin account.

May I have to install a additional tool?

regards Silvio

serdenberger, 2025-01-02:
Hi,
found the export. was an user error.
Just open the menu named "Favorite" and scroll down.

regards Silvio

salfter, 2025-03-26:
> Hi, found the export. was an user error. Just open the menu named "Favorite" **_and scroll down_**.

I just spent more time than I should've bashing my head against a wall until I found this reply.  I'd clicked on "Favorite" and only saw a handful of items.  It never occurred to me that this was a scrollable list, as the scrollbar is located on the opposite side of the screen from the menu items when you have it in a maximized window on a large-enough display.  This impedes discovery of features in the program.

closed 2025-09-07 by jbtronics