Part-DB/Part-DB-server #790: Importing part from LCSC no longer possible with V1.14.5

mkne, 2024-12-06:
**Describe the bug**
After updating the docker image to V1.14.5 from V1.14.4 importing parts from Information Provider LCSC does no longer work and results in a Error 500 message. Deleting the cache with "sudo docker exec --user=www-data partdb php bin/console cache:clear" didn't help. This happens for different users and I made sure that we logged out and logged in after the update of Part-DB.

**To Reproduce**
- Open the page to import a part using an Information Provider (http://partdb:8080/de/tools/info_providers/search)
- Choose LCSC as Information Provider
- Enter the part number "C253558" in the search field
- Click the Search button

**Expected behavior**
Results page with the parts found at LCSC.com

**Screenshots**
[image]

**Server Side**
- Part-DB Version: V1.14.5
- PHP Version: 8.3.14
- Symfony version: 6.4.16
- Database Server SQLite

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Chrome, Edge
 - Version Chrome 131.0.6778.109

**Additional context**
#0 /var/www/html/var/cache/docker/ContainerGsNYtPT/App_KernelDockerContainer.php(201): ContainerGsNYtPT\\getAndroidSafetyNetAttestationStatementSupportService::do()
#1 /var/www/html/var/cache/docker/ContainerGsNYtPT/App_KernelDockerContainer.php(932): ContainerGsNYtPT\\App_KernelDockerContainer->load()
#2 /var/www/html/var/cache/docker/ContainerGsNYtPT/App_KernelDockerContainer.php(2497): ContainerGsNYtPT\\App_KernelDockerContainer::getAttestationStatementSupportManagerService()
#3 /var/www/html/var/cache/docker/ContainerGsNYtPT/App_KernelDockerContainer.php(1236): ContainerGsNYtPT\\App_KernelDockerContainer::getSerializerService()
#4 /var/www/html/var/cache/docker/ContainerGsNYtPT/App_KernelDockerContainer.php(393): ContainerGsNYtPT\\App_KernelDockerContainer::getApiPlatform_Listener_Request_DeserializeService()
#5 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(257): ContainerGsNYtPT\\App_KernelDockerContainer::ContainerGsNYtPT\\{closure}()
#6 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
#7 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
#8 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(157): Symfony\\Component\\EventDispatcher\\EventDispatcher->dispatch()
#9 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
#10 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#11 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#12 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#13 /var/www/html/public/index.php(5): require_once()
#14 {main}
"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}
[2024-12-06T06:57:18.085624+00:00] cache.INFO: Lock acquired, now computing item "search_lcsc_C253558" {"key":"search_lcsc_C253558"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}
[2024-12-06T06:57:18.086136+00:00] http_client.INFO: Request: "GET https://wmsc.lcsc.com/ftps/wm/search/global?keyword=C253558" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}
[2024-12-06T06:57:18.210381+00:00] http_client.INFO: Response: "200 https://wmsc.lcsc.com/ftps/wm/search/global?keyword=C253558" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}
[2024-12-06T06:57:18.210679+00:00] http_client.INFO: Request: "GET https://wmsc.lcsc.com/ftps/wm/product/detail?productCode=C253558" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}
[2024-12-06T06:57:18.310531+00:00] http_client.INFO: Response: "200 https://wmsc.lcsc.com/ftps/wm/product/detail?productCode=C253558" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}
[2024-12-06T06:57:18.310912+00:00] http_client.INFO: Request: "GET https://datasheet.lcsc.com/lcsc/2304140030_Analog-Devices-AD620ARZ-REEL_C253558.pdf" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}
[2024-12-06T06:57:18.476618+00:00] http_client.INFO: Response: "502 https://datasheet.lcsc.com/lcsc/2304140030_Analog-Devices-AD620ARZ-REEL_C253558.pdf" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}
[2024-12-06T06:57:18.477175+00:00] request.CRITICAL: Uncaught PHP Exception Symfony\Component\HttpClient\Exception\ServerException: "HTTP/2 502  returned for "https://datasheet.lcsc.com/lcsc/2304140030_Analog-Devices-AD620ARZ-REEL_C253558.pdf"." at CommonResponseTrait.php line 169 {"exception":"[object] (Symfony\\Component\\HttpClient\\Exception\\ServerException(code: 502): HTTP/2 502  returned for \"https://datasheet.lcsc.com/lcsc/2304140030_Analog-Devices-AD620ARZ-REEL_C253558.pdf\". at /var/www/html/vendor/symfony/http-client/Response/CommonResponseTrait.php:169)
[stacktrace]
#0 /var/www/html/vendor/symfony/http-client/Response/CommonResponseTrait.php(47): Symfony\\Component\\HttpClient\\Response\\CurlResponse->checkStatusCode()
#1 /var/www/html/vendor/symfony/http-client/Response/CurlResponse.php(229): Symfony\\Component\\HttpClient\\Response\\CurlResponse->doGetContent()
#2 /var/www/html/src/Services/InfoProviderSystem/Providers/LCSCProvider.php(105): Symfony\\Component\\HttpClient\\Response\\CurlResponse->getContent()
#3 /var/www/html/src/Services/InfoProviderSystem/Providers/LCSCProvider.php(298): App\\Services\\InfoProviderSystem\\Providers\\LCSCProvider->getRealDatasheetUrl()
#4 /var/www/html/src/Services/InfoProviderSystem/Providers/LCSCProvider.php(212): App\\Services\\InfoProviderSystem\\Providers\\LCSCProvider->getProductDatasheets()
#5 /var/www/html/src/Services/InfoProviderSystem/Providers/LCSCProvider.php(90): App\\Services\\InfoProviderSystem\\Providers\\LCSCProvider->getPartDetail()
#6 /var/www/html/src/Services/InfoProviderSystem/Providers/LCSCProvider.php(143): App\\Services\\InfoProviderSystem\\Providers\\LCSCProvider->queryDetail()
#7 /var/www/html/src/Services/InfoProviderSystem/Providers/LCSCProvider.php(336): App\\Services\\InfoProviderSystem\\Providers\\LCSCProvider->queryByTerm()
#8 /var/www/html/src/Services/InfoProviderSystem/PartInfoRetriever.php(82): App\\Services\\InfoProviderSystem\\Providers\\LCSCProvider->searchByKeyword()
#9 /var/www/html/vendor/symfony/cache/LockRegistry.php(111): App\\Services\\InfoProviderSystem\\PartInfoRetriever->App\\Services\\InfoProviderSystem\\{closure}()
#10 /var/www/html/vendor/symfony/cache/Traits/ContractsTrait.php(102): Symfony\\Component\\Cache\\LockRegistry::compute()
#11 /var/www/html/vendor/symfony/cache-contracts/CacheTrait.php(64): Symfony\\Component\\Cache\\Adapter\\AbstractAdapter->Symfony\\Component\\Cache\\Traits\\{closure}()
[... cut ...]
#20 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#21 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#22 /var/www/html/public/index.php(5): require_once('...')
#23 {main}
"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"MyUser"},"url":"/de/tools/info_providers/search","ip":"x.x.x.x","http_method":"POST","server":"172.16.0.56","referrer":"http://172.16.0.56:8080/de/tools/info_providers/search"}

vvekic, 2024-12-08:
It would seem LCSC responds with 502, regardless of request header.
Looking at the LCSC page, the datasheet URL differs from the one in the JSON returned by /ftps/wm/product/detail.

Example: `https://datasheet.lcsc.com/lcsc/2304140030_Analog-Devices-AD620ARZ-REEL_C253558.pdf` in JSON, but link on page is `https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_Analog-Devices-AD620ARZ-REEL_C253558.pdf`

Maybe this simple rewrite can be implemented in the LCSCProvider.

vvekic, 2024-12-14:
Tested again, and this should be fixed.

nazdridoy, 2024-12-19:
not fixed, Importing part from LCSC is still broken

vvekic, 2024-12-24:
@nazdridoy can you give an example? it's worked for me, i've added dozens of LCSC parts. maybe you ran into one that has the URL broken in a new way

nazdridoy, 2024-12-25:
@vvekic NE555P

[image]
[image]

mkne, 2024-12-26:
@nazdridoy Is the IP address of your part-db server really "0.0.0.0"? This is not a valid host IP address, maybe that is your problem?

vvekic, 2024-12-26:
@nazdridoy it's definitely an issue on your part, unrelated to the datasheet URL issue.
I can find and add the exact same part you mention.

[image]

nazdridoy, 2024-12-27:
@vvekic i don't think i have misconfigured part-db

[image]

[image]

vvekic, 2024-12-27:
For the actual reason of 500 we'd have to see what appears in the seever logs when you try LCSC part search. My guess is it has nothing to do with this particular issue.

nazdridoy, 2024-12-27:
the title of this issue is : "Importing part from LCSC no longer possible with V1.14.5 " and the actual cause of this issue haven't yet been identified. why do you think
> it has nothing to do with this particular issue.

here is the log: 
[report.log](https://github.com/user-attachments/files/18258431/report.log)

vvekic, 2024-12-27:
I don't think you're running the code with the fix.
Your log (line 96) indicates a failure at line #105 during the call to $response->getContent().

However, after commit 0f92a69b038dda0418e080a458c35533bfec5603 (my LCSC fix), $response->getContent() is on line 108, not 105. https://github.com/Part-DB/Part-DB-server/blob/master/src/Services/InfoProviderSystem/Providers/LCSCProvider.php#L108

I'm guessing you're still running the release 1.14.5, and not master.

If you want to see the fix, run master or wait for the maintainer to tag a new release, then upgrade.

vvekic, 2024-12-29:
@nazdridoy or anyone else - are you still able to reproduce the error on master?

If, like me, you aren't, I suggest a @jbtronics to close this issue.

nazdridoy, 2024-12-30:
I'd suggest keeping this issue open until a new release is published, as this issue is still reproducible in the latest release.

nonbinary-duck, 2024-12-31:
> [@]nazdridoy or anyone else - are you still able to reproduce the error on master?

LCSC part provider working again on master for me. Thank you!

[image]

mkne, 2024-12-31:
I would close this issue but as long as it is not in the release docker image I cannot test it. Or is there a docker image for the master as well?

jbtronics, 2024-12-31:
> Or is there a docker image for the master as well?

The master or edge tag of jbtronics/part-db1.
So something like jbtronics/part-db1:edge

grzegorzwozny, 2025-01-05:
I also encountered an **Internal Server Error (Status 500)** when trying to use LCSC. The fix in [Rewrite LCSC datasheet URL #795](https://github.com/Part-DB/Part-DB-server/pull/795) resolved the issue on my end.

closed 2025-01-05 by jbtronics