Part-DB/Part-DB-server #792: Add Serial Number (SN) to components

Alefriends97, 2024-12-06:
Hi,

is it possible to add serial number tracking to parts/components?

i checked the "parameters" option of a part but it doesn't seem suitable

jbtronics, 2024-12-06:
You mean like an SN that is unique for each instance of a part?

Part-DB normally assumes that all instances of a part are the same and interchangeable.

If you wanna track individual things you can maybe use individual part lots for each instance. You can use the comment field there to add additional info there (like a SN)

Dreamtrailtv, 2025-01-09:
I would also very much welcome this.
For example, we have parts where we need to know what the serial number and/or Mac address of a particular part is in our warehouse. This is a safety stock where goods need to be stored where we need to know the exact products at all times

mkne, 2025-02-06:
Without reading the comment from @jbtronics I have done it the way he proposed. So, I think it's a least a bit intuitive. I did it the following way:
1. Create a storage location for every workplace/desk where a measuring tool (multimeter) can be (in our case: Arbeitsplatz -> Workers Name)
2. Add the part lots to the part and assign the storage location
3. Enter the serial number and internal measuring tool number as description of the part
4. optional: add expiration date to track the need of calibration

[image]

However, the major problem with this solution is that **the search does not find the part** or instance if I search for the serial number.
Another point is how to handle for example project build instances. Adding serial numbers the way described above to the my assembled pcbs (that are project builds) is not very practical as all instances have the same storage location.

adamoskaranikas, 2025-04-24:
I second this functionality.

I think having a built in way to make a part have unique serial number for each part instance would be useful. For me, the specific use case would be if I have an instance of a project build, I'd want to know the serial number of each complex part that is used, so any flaws or hardware errors can be tracked to specific components. Or If the serial number property was exposed, then one could even tie specific parts together through the Associated Parts functionality. At the moment, using the part lots as a workaround is not ideal, as it can get clunky in the Parts Overview UI.

mkne, 2025-05-01:
> I second this functionality.
>
> I think having a built in way to make a part have unique serial number for each part instance would be useful. For me, the specific use case would be if I have an [...]

I agree that a proper serial number property would be great (for example for parts with a MAC address) But how would you model that into the current data structure? Because you would need multiple instances of parts. I would love to have serial numbers for build instances of projects to manage/track our prototypes. Maybe there's something already hidden in Part-DB that would allow that use case? Just checked the Build feature in Projects and it actually just adds the amount of build instances of the project as a stock position.