Part-DB/Part-DB-server #796: PartDB on Raspberry Pi

MatthewAguiar, 2024-12-10:
**Describe the bug**
I am trying to set up PartDB on a Raspberry Pi 5 using the Apache2 direct installation. After following the instructions for direct installation closely a handful of times, my PC can only fetch assets like the favicon.ico. If I try and load the main index.php, then it can't seem to reach that resource. One image below shows, "This site can't be reached", while the other image shows the favicon.ico being fetched properly. I tried to fetch the favicon just to prove Apache is actually active and serving files.

I'm trying to find /var/www/partdb/var/log/prod.log to provide more info but it doesn't seem to exist.

Interestingly, I'm seeing entries in the /var/log/apache2/partdb_access.log when I try to fetch the favicon but no entries when trying to fetch the main index.php. So it seems as though the request isn't getting through.

I did disable the default apache site by the way so it should locate it.

**To Reproduce**
Steps to reproduce the behavior:
[https://docs.part-db.de/installation/installation_guide-debian.html](https://docs.part-db.de/installation/installation_guide-debian.html)

**Expected behavior**
The app should at least show a login screen correct?

**Screenshots**
[image]
[image]

**Server Side**
- Part-DB Version: Part-DB 1.14.5
- PHP Version: PHP 8.1
- Database Server MariaDB 15.1

**Desktop**
 - OS: Windows 10
 - Browser: Chrome

MatthewAguiar, 2024-12-10:
Follow up. It seems like if I change the ownership of the partdb folder from www-data to root, I at least get a 500 error to return.
`sudo chown -R root:root /var/www/partdb`

[image]

jbtronics, 2024-12-10:
If you get a connection refused message, then there is an issue with your webserver and or PHP config, not Part-DB (as it did not get called at that point yet). And if it not was called yet you do not get a log by Part-DB.

You should look into the global apache error log to check if there are issues.

The second screenshot was generated by symfony. You should get logs from Part-DB explaining the internal error more. But this will not work with root as owner of the folder, as then the webserver cannot create the log file. (And no cache, which is probably the reason for the 500 error).

MatthewAguiar, 2024-12-10:
Oh. OK. I just changed the owner back to www-data and somehow it's working now. Strange. Thank you, though.

closed 2024-12-10 by MatthewAguiar