Part-DB/Part-DB-server #798: Add different project based databases

PeterKremsner, 2024-12-11:
We are currently looking for a possibility to create different databases for different projects. The reason why we would like to have it that way is that each project has its own parts and parts from one project cannot be transfered over to another project. Within a project we have multiple different pcbs, a dedicated storage and so on, so they are completly independet like multible instances of part-db

I want to include this feature into part-db and so im currently searching for possibilities how to achieve this. 

For the UI i thought of a dropdown menu in the top menu where the active database can be selected, the rest can stay the same.

I wanted to ask for some informations or clues how i can integrate this mechanism into the code, like is there a database interface class which i can modify so it automatically loads parts, projects and so on from another database or alter the SQL in a way so that only parts with a specific value in a column are used?

Can i change this in one file or do i need to modify each query?

The alternative would be to use different part-dbs however the users are not bound to a specific project and are global. Also global storage that should be visible in every project could be needed in a later stage, so a single instance of part-db would be better.

jbtronics, 2024-12-11:
I don't think it's a good idea to switch between different databases. Symfony's and doctrine's architecture are built around that the assumption that the database is static and global. If you switch between databases regularly and even different for users, you will break stuff. Especially with caching mechanism and similar, you will run into trouble. Also, it will be very difficult to have shared user base. Changing everything would be a huge amount of work.

Using one single database, where you apply filters to only show certain parts, is most certainly the better approach here. Maybe one could utilize [doctrines filter mechanism](https://www.doctrine-project.org/projects/doctrine-orm/en/3.3/reference/filters.html) for that, but i have never actually used it. And you would still need to change every caching mechanism according to that. So that would still be a lot of work, and I would assume that this will make maintainability pretty difficult.

Probably it's easier to just use docker or another container solution to quickly create a new Part-DB instance for every new project, and use SSO to get a shared user authentication between the instances...

PeterKremsner, 2024-12-22:
I've implemented some changes that will filter the components based on an environment.

https://github.com/Part-DB/Part-DB-server/compare/master...PeterKremsner:Part-DB-server:part_environments

I currently try to show the environment of a part in the overview table that gets displayed when you show the parts in an category. However i am not able to find the twig template that is used to create this table. Do you have a hint?

The changes on the cache should not be too drastic since the part filter is always called even on cached database entries maybe the change would be just adding the environment variable to the cache. However i didnt look into this specifically. If the environment is zero the environment is global so not using the environment at all is also possible if one doesn't need it.

Since the part id is, of course, still unique to a part, there are also no changes in editing the part and so on necessary. The filter just doesnt show all parts that dont match the environment current environment.