Part-DB/Part-DB-server #803: Internal Server Error when creating new Part

BangerTech, 2024-12-12:
**Describe the bug**
I just set up a new freshly installed docker container.
migrated the database from default to mariadb/mysql.
setting up user and attachment types and categories is no problem, but if i create a new part and click on save changes this error occurs.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'create new part'
2. Click on 'save changes'
3. Scroll down to '....'
4. See error

**Expected behavior**
the ability to create a new part.

**Screenshots**
[image]

**Server Side**
- Part-DB Version: latest
- PHP Version: latest docker & docker-compose
- Database Server latest mariadb database
all in a seperate container which are all on the same docker network

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser chrome,thorium, safari
 - Version [e.g. 22]

**Additional context**
`services:

  partdb:
    container_name: partdb
    # By default Part-DB will be running under Port 8080, you can change it here
    ports:
      - '8034:80'
    volumes:
      # By default
      - ./uploads:/var/www/html/uploads
      - ./public_media:/var/www/html/public/media
      - ./db:/var/www/html/var/db
     # - ./middlewares.yml:/etc/traefik/middlewares.yml
    restart: unless-stopped
    image: jbtronics/part-db1:latest
  #  depends_on:
  #    - mariadb
    environment:
      # Put SQLite database in our mapped folder. You can configure some other kind of database here too.
     # - DATABASE_URL=sqlite:///%kernel.project_dir%/var/db/app.db
      - DATABASE_URL=mysql://partdb:root@mariadb:3306/partdb
      # In docker env logs will be redirected to stderr
      - APP_ENV=docker

      # You can configure Part-DB using environment variables
      # Below you can find the most essential ones predefined
      # However you can add any other environment configuration you want here
      # See .env file for all available options or https://docs.part-db.de/configuration.html
      # !!! Do not use quotes around the values, as they will be interpreted as part of the value and this will lead to errors !!!

      # The language to use serverwide as default (en, de, ru, etc.)
      - DEFAULT_LANG=de
      # The default timezone to use serverwide (e.g. Europe/Berlin)
      - DEFAULT_TIMEZONE=Europe/Berlin
      # The currency that is used inside the DB (and is assumed when no currency is set). This can not be changed later, so be sure to set it the currency used in your country
      - BASE_CURRENCY=EUR
      # The name of this installation. This will be shown as title in the browser and in the header of the website
      - INSTANCE_NAME=Part-DB

      # Allow users to download attachments to the server by providing an URL
      # This could be a potential security issue, as the user can retrieve any file the server has access to (via internet)
      - ALLOW_ATTACHMENT_DOWNLOADS=0
      # Use gravatars for user avatars, when user has no own avatar defined
      - USE_GRAVATAR=0
      - PROVIDER_MOUSER_KEY=233b86cd-8c7a-4299-a722-e1616f5b3cf2
      - DEFAULT_URI=https://parts.server24-7.net
      # Override value if you want to show a given text on homepage.
      # When this is empty the content of config/banner.md is used as banner
      #- BANNER=This is a test banner<br>with a line break
#      - CONTENT_SECURITY_POLICY="default-src 'self'; connect-src 'self' https://nodered.server24-7.net;"    
      # If you use a reverse proxy in front of Part-DB, you must configure the trusted proxies IP addresses here (see reverse proxy documentation for more information):
      - TRUSTED_PROXIES=192.168.2.55,192.168.2.182,192.168.2.1,192.168.2.83  
    labels:
      #- traefik.http.routers.parts.middlewares=csp-header@redis
      - traefik.http.middlewares.csp-header.headers.customResponseHeaders.Content-Security-Policy=default-src 'self'; connect-src 'self' https://nodered.server24-7.net
      - traefik.http.routers.parts.middlewares=csp-header@redis
      - traefik.enable=true
      - traefik.http.routers.parts.rule=Host(`parts.server24-7.net`)
      - traefik.http.routers.parts.tls=true
      - traefik.http.routers.parts.tls.certresolver=cloudflare
      - traefik.http.services.parts.loadbalancer.server.scheme=http
      - traefik.http.services.parts.loadbalancer.server.port=8034
      #- traefik.http.routers.parts.entrypoints=websecure

    networks:
      - proxy

networks:
  proxy:
    external: true`

`services:
  mariadb:
    image: mariadb:latest
    container_name: mariadb
    restart: unless-stopped
    environment:
      - MYSQL_ROOT_PASSWORD=root
      - MYSQL_DATABASE=partdb
      - MYSQL_USER=partdb
      - MYSQL_PASSWORD=root
    volumes:
      - ./mariadb-data:/var/lib/mysql
    ports:
      - "3306:3306"
    networks:
      - proxy

networks:
  proxy:
    external: true
`

jbtronics, 2024-12-12:
you should check the docker logs

BangerTech, 2024-12-12:
partdb  | 192.168.2.182 - - [12/Dec/2024:18:52:24 +0000] "GET /de/part/new?category=3 HTTP/1.1" 200 15713 "https://parts.server24-7.net/de/category/3/parts" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/124.0.0.0 Safari/537.36"
partdb  | [2024-12-12T18:52:27.705839+00:00] request.INFO: Matched route "part_new". {"route":"part_new","route_parameters":{"_route":"part_new","_controller":"App\\Controller\\PartController::new","_locale":"de"},"request_uri":"https://parts.server24-7.net/de/part/new?category=3","method":"POST"} {"token":null,"url":"/de/part/new?category=3","ip":"192.168.2.11","http_method":"POST","server":"parts.server24-7.net","referrer":"https://parts.server24-7.net/de/part/new?category=3"}
partdb  | [2024-12-12T18:52:27.708305+00:00] deprecation.INFO: User Deprecated: Since api-platform/core 3.3: Use a "ApiPlatform\State\ProviderInterface" as first argument in "ApiPlatform\Symfony\EventListener\QueryParameterValidateListener" instead of "ApiPlatform\ParameterValidator\ParameterValidator". {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since api-platform/core 3.3: Use a \"ApiPlatform\\State\\ProviderInterface\" as first argument in \"ApiPlatform\\Symfony\\EventListener\\QueryParameterValidateListener\" instead of \"ApiPlatform\\ParameterValidator\\ParameterValidator\". at /var/www/html/vendor/api-platform/core/src/Symfony/EventListener/QueryParameterValidateListener.php:47)
partdb  | [stacktrace]
partdb  | #0 /var/www/html/var/cache/docker/ContainerOveKFnc/App_KernelDockerContainer.php(1309): ApiPlatform\\Symfony\\EventListener\\QueryParameterValidateListener->__construct()
partdb  | #1 /var/www/html/var/cache/docker/ContainerOveKFnc/App_KernelDockerContainer.php(398): ContainerOveKFnc\\App_KernelDockerContainer::getApiPlatform_Listener_View_ValidateQueryParametersService()
partdb  | #2 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(257): ContainerOveKFnc\\App_KernelDockerContainer::ContainerOveKFnc\\{closure}()
partdb  | #3 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
partdb  | #4 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
partdb  | #5 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(157): Symfony\\Component\\EventDispatcher\\EventDispatcher->dispatch()
partdb  | #6 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
partdb  | #7 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
partdb  | #8 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
partdb  | #9 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
partdb  | #10 /var/www/html/public/index.php(5): require_once()
partdb  | #11 {main}
partdb  | "} {"token":null,"url":"/de/part/new?category=3","ip":"192.168.2.11","http_method":"POST","server":"parts.server24-7.net","referrer":"https://parts.server24-7.net/de/part/new?category=3"}
partdb  | [2024-12-12T18:52:27.710988+00:00] security.DEBUG: Read existing security token from the session. {"key":"_security_main","token_class":"Symfony\\Component\\Security\\Core\\Authentication\\Token\\UsernamePasswordToken"} {"token":null,"url":"/de/part/new?category=3","ip":"192.168.2.11","http_method":"POST","server":"parts.server24-7.net","referrer":"https://parts.server24-7.net/de/part/new?category=3"}
partdb  | [2024-12-12T18:52:27.713365+00:00] deprecation.INFO: User Deprecated: Support for MySQL < 8 is deprecated and will be removed in DBAL 5 (AbstractMySQLDriver.php:70 called by AbstractDriverMiddleware.php:32, https://github.com/doctrine/dbal/pull/6343, package doctrine/dbal) {"exception":"[object] (ErrorException(code: 0): User Deprecated: Support for MySQL < 8 is deprecated and will be removed in DBAL 5 (AbstractMySQLDriver.php:70 called by AbstractDriverMiddleware.php:32, https://github.com/doctrine/dbal/pull/6343, package doctrine/dbal) at /var/www/html/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php:210)
partdb  | [stacktrace]
partdb  | #0 /var/www/html/vendor/doctrine/deprecations/lib/Doctrine/Deprecations/Deprecation.php(108): Doctrine\\Deprecations\\Deprecation::delegateTriggerToBackend()
partdb  | #1 /var/www/html/vendor/doctrine/dbal/src/Driver/AbstractMySQLDriver.php(70): Doctrine\\Deprecations\\Deprecation::trigger()
partdb  | #2 /var/www/html/vendor/doctrine/dbal/src/Driver/Middleware/AbstractDriverMiddleware.php(32): Doctrine\\DBAL\\Driver\\AbstractMySQLDriver->getDatabasePlatform()
partdb  | #3 /var/www/html/vendor/doctrine/dbal/src/Driver/Middleware/AbstractDriverMiddleware.php(32): Doctrine\\DBAL\\Driver\\Middleware\\AbstractDriverMiddleware->getDatabasePlatform()
partdb  | #4 /var/www/html/vendor/doctrine/dbal/src/Driver/Middleware/AbstractDriverMiddleware.php(32): Doctrine\\DBAL\\Driver\\Middleware\\AbstractDriverMiddleware->getDatabasePlatform()
partdb  | #5 /var/www/html/vendor/doctrine/doctrine-bundle/src/ConnectionFactory.php(122): Doctrine\\DBAL\\Driver\\Middleware\\AbstractDriverMiddleware->getDatabasePlatform()
[... cut ...] t/new?category=3","ip":"192.168.2.11","http_method":"POST","server":"parts.server24-7.net","referrer":"https://parts.server24-7.net/de/part/new?category=3"}
partdb  | 192.168.2.182 - - [12/Dec/2024:18:52:27 +0000] "POST /de/part/new?category=3 HTTP/1.1" 500 4570 "https://parts.server24-7.net/de/part/new?category=3" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/124.0.0.0 Safari/537.36"
partdb  | - -  12/Dec/2024:18:52:27 +0000 "POST /index.php" 500

BangerTech, 2024-12-12:
sorry for the bad formatting ;-) found the error in one of my Triggers

closed 2024-12-12 by BangerTech