Part-DB/Part-DB-server #806: Missing ability to save Label profile

alextrical, 2024-12-18:
**Describe the bug**
After creating a label through the Label Generator, I cannot locate a method to save the configuration for later use, only the "Load profile", which implies there should be a way to save the profile in the first place

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Tools > Label generator'
2. Configure a label as you would like to print
3. Click on 'Profiles'
4. See only the option to 'Load profile', but not save

**Expected behavior**
I expect after configuring the label layout, to be able to save the layout profile for future use. This is expecially useful when there are multiple label formats required for different component storage locations

**Screenshots**
[image]

**Server Side**
- Part-DB Version: 1.14.5 
- PHP Version: 8.3.14
- Database Server: 10.11.10-MariaDB

**Desktop (please complete the following information):**
 - OS: MacOS 14.6.1
 - Browser Firefox
 - Version 133.0.3

alextrical, 2024-12-19:
Just spotted this uder the edit menu. It looks like this is the method i should be using. A link on the print page, or some description in the documentation would be useful to new users. I will look at adding a comment to the docs as a PR over the next week
[image]

jbtronics, 2025-01-05:
Now there is a button to directly save a new label profile directly from the label profile.

closed 2025-01-05 by jbtronics