Part-DB/Part-DB-server #807: Importing part from octopart doesn't work

nazdridoy, 2024-12-19:
**Describe the bug**

 Importing parts from Information Provider octopart does not work and results in  "No results found at the selected providers! Check your search term or try to choose additional providers.".

**To Reproduce**

Open the page to import a part using an Information Provider (http://partdb:8080/de/tools/info_providers/search)
- Choose octopart as Information Provider
- Enter the part number "rp2040" in the search field
- Click the Search button

**Expected behavior**
Results page with the parts found at octopart

**Screenshots**
octopart not working
[image]

TME is working
[image]

**Server Side**
- Part-DB Version: 1.14.5
- PHP Version: 8.3.14 (SAPI: fpm-fcgi)
- Database Server: qlite 3.40.1

**Desktop (please complete the following information):**
 - OS: Archlinux
 - Browser : Brave
 - Version : Version 1.73.97 Chromium: 131.0.6778.108 (Official Build) (64-bit)

**Additional context**
[image]

LCSC is also broken: #790

mkne, 2024-12-19:
Did you check if you haven't exceeded your Nexar usage limit? You should see it in the dashboard when you login at https://portal.nexar.com/ I was able to add a part using Octopart just yesterday.

nazdridoy, 2024-12-19:
this is a brand new account. 
[image]

according to the wiki:
>create a new application on the Nexar Portal. The name does not matter, but it is important that the application has access to the “Supply” scope. In the Authorization tab, you will find the client ID and client secret, which you have to put in the Part-DB env configuration (see below).
Please note that the Nexar API in the free plan is limited to 1000 results per month.

mkne, 2024-12-26:
You've exceeded your limit, hence it is displaying 100% Usage. But I don't understand why your limit is 0 mine is for example 1000 and if I have used 800 the percentage show 80% used. But this is not an issue of part-db but of Nexar, you should consult with the Nexar support or upgrade your plan. And close this issue.

nazdridoy, 2024-12-27:
i also think Nexar is the problem...  closing this issue..

closed 2024-12-27 by nazdridoy

gknit, 2025-03-15:
You need to register with Nexar and then choose a plan (the free one called "Evaluation" has 1000 monthly results). If you just create an app without registering for a plan, you will have a limit of 0.