Part-DB/Part-DB-server #819: php bin/console commands not working

BS-Foster, 2025-01-04:
**Describe the bug**

I'm unable to use any php commands with bin/console. it returns the error "Could not open input file: bin/console". I can't check version of or update part-db. Not sure if the is related to the issue, but im also unable to open/interact with any of the drop down menus with the (^) icon on the client side. 

**To Reproduce**
Steps to reproduce the behavior:
1. In the part-db installation, type "php bin/console partdb:version" or any command with "php bin/console partdb:". 

**Expected behavior**

I'm expecting it to return the version number of my part-db installation or update my installation.

**Screenshots**

N/A

**Server Side**
- Part-DB Version: N/A (this is part of my issue)
- PHP Version: 8.2.26
- Database Server: PostgreSQL version 15.10

**Desktop (please complete the following information):**
 - OS: [Windows 11
 - Browser: Chrome
 - Version: 131.0.6778.205

**Smartphone (please complete the following information):**
- N/A

**Additional context**

When I was installing part-db through the script, and after it was done, I went to the i.p. and it showed the Apache2 webpage showing the the Apache server is running. I had to reboot my part-db install, and afterwards it showed the part-db home page. Not sure if this is also related but I ran into a few issues just getting to this point. I'm now having the issues(s) stated above.

jbtronics, 2025-01-04:
You are probably in the wrong directory. You must be in Part-DBs root folder for this to properly work

BS-Foster, 2025-01-05:
@jbtronics Ah, yep. Just to confirm it is /opt/partdb, right? Thank ya

jbtronics, 2025-01-05:
That depends on your installation path. I would expect it to be in a webserver folder, but that depends  on your configuration.

closed 2025-01-05 by jbtronics