Part-DB/Part-DB-server #820: Problem using the Mouser provider: Unknown response format

liborw, 2025-01-05:
**Describe the bug**
Added the Mouser API key, but get `Internal Server Error (Status 500)`, tried to clear the cache as suggested with no success. Went into logs and this is the part that is related to mouser:
```
partdb           | [2025-01-05T14:18:10.848723+00:00] cache.INFO: Lock acquired, now computing item "search_mouser_test" {"key":"search_mouser_test"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"172.26.0.20","http_method":"POST","server":"partdb.xxx.xx","referrer":"https://partdb.xxx.xx/en/tools/info_providers/search"}
partdb           | [2025-01-05T14:18:10.849072+00:00] http_client.INFO: Request: "POST https://api.mouser.com/api/v2/search/keyword?apiKey=29b4bab4562b30291cb41e5e991dde2e3ac2a7f5d2ee307f51" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"172.26.0.20","http_method":"POST","server":"partdb.xxx.xx","referrer":"https://partdb.xxx.xx/en/tools/info_providers/search"}
partdb           | [2025-01-05T14:18:11.613668+00:00] http_client.INFO: Response: "200 https://api.mouser.com/api/v2/search/keyword?apiKey=29b4bab4562b30291cb41e5e991dde2e3ac2a7f5d2ee307f51" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"172.26.0.20","http_method":"POST","server":"partdb.xxx.xx","referrer":"https://partdb.xxx.xx/en/tools/info_providers/search"}
partdb           | [2025-01-05T14:18:11.615105+00:00] request.CRITICAL: Uncaught PHP Exception RuntimeException: "Unknown response format" at MouserProvider.php line 208 {"exception":"[object] (RuntimeException(code: 0): Unknown response format at /var/www/html/src/Services/InfoProviderSystem/Providers/MouserProvider.php:208)
partdb           | [stacktrace]
partdb           | #0 /var/www/html/src/Services/InfoProviderSystem/Providers/MouserProvider.php(135): App\\Services\\InfoProviderSystem\\Providers\\MouserProvider->responseToDTOArray()
partdb           | #1 /var/www/html/src/Services/InfoProviderSystem/PartInfoRetriever.php(82): App\\Services\\InfoProviderSystem\\Providers\\MouserProvider->searchByKeyword()
```

**Server Side**
- Part-DB Version: 1.15.0-dev (docker edge)
- PHP Version: 8.3.15 (SAPI: fpm-fcgi)
- Database Server MySql 	8.0.40

**Desktop:**
 - OS: Archlinux
 - Browser: qutebrowser
 - Version: 3.4.0

**Additional context**
Using docker and Nginx reverse proxy.

jbtronics, 2025-01-05:
I would assume that something with your mouser configuration is wrong, or the mouser API key is somehow invalid. You should check this.

In the next Part-DB version (1.15) the error log should include more info about this.

romihsan, 2025-01-17:
I am trying to evaluate Part-DB and have therefore setup a docker contain as per the instructions in docs.part-db.de
To keep it simple, I just copied the 1st docker-compose.yml example on the "Installing Using Docker" page.

Part-DB is functioning, and I was able to add a part manually.

I ran into the same issue as the OP when trying to use my mouser API key. I get the Internal Server Error 500 message when I try to "Create parts from info provider", and the log messages look similar to the OP's logs.

Part-DB version 1.15.1

jbtronics, 2025-01-17:
The Part-DB 1.15 versions should include a more detailed error message, containing the exact errronous response.

Can you post the relevant log entries here?

romihsan, 2025-01-17:
My log output is below:

```
[] cache.INFO: Lock acquired, now computing item "search_mouser_maxm17572" {"key":"search_mouser_maxm17572"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"172.18.0.1","http_method":"POST","server":"localhost","referrer":"http://localhost:8080/en/tools/info_providers/search"}
[] http_client.INFO: Request: "POST https://api.mouser.com/api/v2/search/keyword?apiKey=dc5adb6f-276f-4ebc-a700-4e91460b412f" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"172.18.0.1","http_method":"POST","server":"localhost","referrer":"http://localhost:8080/en/tools/info_providers/search"}
[] http_client.INFO: Response: "200 https://api.mouser.com/api/v2/search/keyword?apiKey=dc5adb6f-276f-4ebc-a700-4e91460b412f" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"172.18.0.1","http_method":"POST","server":"localhost","referrer":"http://localhost:8080/en/tools/info_providers/search"}
[] request.CRITICAL: Uncaught PHP Exception RuntimeException: "Unknown response format: {"Errors":[{"Id":0,"Code":"Invalid","Message":"Invalid unique identifier.","ResourceKey":"InvalidIdentifier","ResourceFormatString":null,"ResourceFormatString2":null,"PropertyName":"API Key"}],"SearchResults":null}" at MouserProvider.php line 208 {"exception":"[object] (RuntimeException(code: 0): Unknown response format: {\"Errors\":[{\"Id\":0,\"Code\":\"Invalid\",\"Message\":\"Invalid unique identifier.\",\"ResourceKey\":\"InvalidIdentifier\",\"ResourceFormatString\":null,\"ResourceFormatString2\":null,\"PropertyName\":\"API Key\"}],\"SearchResults\":null} at /var/www/html/src/Services/InfoProviderSystem/Providers/MouserProvider.php:208)
```

jbtronics, 2025-01-17:
I would interpret the error message, that mouser does not recognize your API key. you should check, that it is the correct one.

romihsan, 2025-01-17:
I have checked it already, well at least the beginning and end digits.

A question regarding the .env file. The API key should be contained inside ' ' right?

For example:
`PROVIDER_MOUSER_KEY='dc5adb6f-..........-460b412f'`

romihsan, 2025-01-17:
I do have a warning message on the Mouser API page telling me that I do not have access to certain features:

[image]

But I was assuming that I should still be able to search their databases. Is that assumption incorrect?

jbtronics, 2025-01-17:
You must sign up to the Search API using this form: https://eu.mouser.com/api-search/

I would assume that access to search API does not work with the API key, which you can generate from under the "My Account" page (that is most likely just for the account related API endpoints).

romihsan, 2025-01-17:
Ahhhh yes!!!
That was it.

Thanks for your help, and please forgive me for having taken up your time.

closed 2025-03-29 by jbtronics