Part-DB/Part-DB-server #822: 500 Internal Server Error The XML file "../config/dbal.xml" is not valid

Pytonballoon810, 2025-01-07:
**Describe the bug**
I was trying to set up backups and ran into this issue. I am running the PartDB via Portainer docker-compose and had no issues until now. I already stopped, deleted, restarted all the containers involved (SQL container and the app container). I also already deleted the container image for the app as the issue does not seem to be related to the My-SQL container. 

**Screenshots**
[image]
[image]

**Versions**
I am using the newest versions of everything. I just pulled the latest docker image

**Docker Compose**
```yaml
version: '3.3'
services:
  partdb:
    container_name: partdb
    ports:
      - '8100:80'
    volumes:
      - /home/partdb/uploads:/var/www/html/uploads
      - /home/partdb/backups:/tmp
      - /home/partdb/public_media:/var/www/html/public/media
      - /home/partdb/db:/var/www/html/var/db
    restart: unless-stopped
    image: jbtronics/part-db1:latest
    depends_on:
      - database
    environment:
      # Replace SECRET_USER_PASSWORD with the value of MYSQL_PASSWORD from below
      - DATABASE_URL=mysql://partdb:@database:3306/partdb
      # In docker env logs will be redirected to stderr
      - APP_ENV=docker

      # You can configure Part-DB using environment variables
      # Below you can find the most essential ones predefined
      # However you can add add any other environment configuration you want here
      # See .env file for all available options or https://docs.part-db.de/configuration.html

      # The language to use serverwide as default (en, de, ru, etc.)
      - DEFAULT_LANG=en
      # The default timezone to use serverwide (e.g. Europe/Berlin)
      - DEFAULT_TIMEZONE=Europe/Berlin
      # - DEFAULT_URI= https://parts.nobble.it
      # The currency that is used inside the DB (and is assumed when no currency is set). This can not be changed later, so be sure to set it the currency used in your country
      - BASE_CURRENCY=EUR
      # The name of this installation. This will be shown as title in the browser and in the header of the website
      - INSTANCE_NAME=Part-DB

      # Allow users to download attachments to the server by providing an URL
      # This could be a potential security issue, as the user can retrieve any file the server has access to (via internet)
      - ALLOW_ATTACHMENT_DOWNLOADS=1
      # Use gravatars for user avatars, when user has no own avatar defined
      - USE_GRAVATAR=1
      - PROVIDER_LCSC_ENABLED=1

      # Override value if you want to show to show a given text on homepage.
      # When this is empty the content of config/banner.md is used as banner
      #- BANNER=This is a test banner<br>with a line break

      - TRUSTED_PROXIES=10.0.0.0/8
    networks: 
      - casa
      
  database:
    container_name: partdb_database
    image: mysql:8.0
    restart: unless-stopped
    command: --default-authentication-plugin=mysql_native_password --log-bin-trust-function-creators=1
    environment:
      # Change this Password
      MYSQL_ROOT_PASSWORD: 
      MYSQL_DATABASE: 
      MYSQL_USER: 
      MYSQL_PASSWORD: 
    # Uncomment the following line if you need to access, your MySQL database from outside of docker (e.g. for debugging), normally you should leave that disabled
    #ports:
    #  - '4306:3306'
    volumes:
      - /home/partdb/mysql:/var/lib/mysql
    networks: 
      - casa
networks:
  casa:
    external: true

```

Pytonballoon810, 2025-01-07:
I tested if the file is even there where it should be by running `docker exec -it partdb ls /var/www/html/vendor/doctrine/doctrine-bundle/config/dbal.xml` I found that it was missing. I then restarted everything again and deleted all mounted directories except the db. I now have the file again. When throwing the sml file in a online validator i found that i get 
```
Errors in the XML document:
		  
schema_reference.4:  Failed to read schema document  'http://symfony.com/schema/dic/services/services-1.0.xsd', because 1)  could not find the document; 2) the document could not be read; 3) the  root element of the document is not <xsd:schema>.
```

I then asked chat gpt and it told me to do `docker exec -it partdb curl -I http://symfony.com/schema/dic/services/services-1.0.xsd` to test if the file is there or something.

I get this back:
```
HTTP/1.1 301 Moved Permanently
Date: Tue, 07 Jan 2025 01:40:13 GMT
Content-Type: text/html
Content-Length: 167
Connection: keep-alive
Cache-Control: max-age=3600
Expires: Tue, 07 Jan 2025 02:40:13 GMT
Location: https://symfony.com/schema/dic/services/services-1.0.xsd
Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=oQJY4N4XG21nlb7TpSqW%2BT12zBDjbosqj2Kt%2BmKW9fTon9I1grX8eclfC3cj1iiptTto7Nx%2FaxvmJHA6fuhWXwew7aPM2OuMXe9yVj7Tkme5u5ly%2Fbxc8ZbrByOC"}],"group":"cf-nel","max_age":604800}
NEL: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
Server: cloudflare
CF-RAY: 8fe03d0ecd44c7e9-DUS
server-timing: cfL4;desc="?proto=TCP&rtt=21761&min_rtt=21761&rtt_var=10880&sent=1&recv=3&lost=0&retrans=0&sent_bytes=0&recv_bytes=112&delivery_rate=0&cwnd=239&unsent_bytes=0&cid=0000000000000000&ts=0&x=0"
```

I then tried to clear the cache by doing `docker exec -u www-data partdb php bin/console cache:clear` but i only get 
```
In XmlFileLoader.php line 486:
                                                                               
  Unable to parse file "/var/www/html/vendor/doctrine/doctrine-bundle/src/Dep  
  endencyInjection/../../config/dbal.xml": The XML file "/var/www/html/vendor  
  /doctrine/doctrine-bundle/src/DependencyInjection/../../config/dbal.xml" is  
   not valid.                                                                  
                                                                               

In XmlUtils.php line 134:
                                                                               
  The XML file "/var/www/html/vendor/doctrine/doctrine-bundle/src/DependencyI  
  njection/../../config/dbal.xml" is not valid.                                
                                                                               

In XmlFileLoader.php line 793:
                                                                       
  Notice: tempnam(): file created in the system's temporary directory 
```

I dont know whether this has anything to do with my network config of the container since i didnt change anything and it just stopped working after testing out the backup function.

I also find it extraordinarily strange strange that the issue persist even after repulling the container image and deleting all files not containing any database data.

Pytonballoon810, 2025-01-07:
I seem to have fixed the issue:

I did the following things to make it work:
- `docker exec -it partdb /bin/bash` to enter the container console
- `chmod 1777 /tmp`
- `exit` exit the container env
- `docker exec -it partdb apt-get update` update packages
- `docker exec -it partdb apt-get install -y nano` install nano in the container
- `docker exec -it partdb nano /var/www/html/vendor/doctrine/doctrine-bundle/config/dbal.xml` edit the broken file throwing the errors
- Replace the first part with the "Moved Permanently" link with this:
```xml
<container xmlns="http://symfony.com/schema/dic/services"
           xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
           xsi:schemaLocation="http://symfony.com/schema/dic/services https://symfony.com/schema/dic/services/services-1.0.xsd">
```

- `docker exec -u www-data partdb php bin/console cache:clear` to clear the application cache.

DONE! I now can use the app again as intended.

Pytonballoon810, 2025-01-07:
If anybody has any Idea what caused this i would be more than interested.

jbtronics, 2025-01-07:
I never encountered such an error and don't quite understand what's happening there.

However, I do not really think that this is related to this file itself. This is part of doctrine itself, it would be very unlikely that it is erronous.
I think this was caused by a non-writeable `/tmp` folder and your `chmod` command fixed that.
At least everything what I found online regarding this error, seems to be related to the `/tmp` directory.

closed 2025-01-12 by Pytonballoon810