Part-DB/Part-DB-server #824: upgrade from 1.14.5 to 1.15.0 not not loading javascript 404

trefase, 2025-01-07:
Bug Description:
 After updating from version 1.14.5 to 1.15.0 on a Debian system, Part-DB no longer loads JavaScript resources, resulting in 404 errors for JavaScript files. The previous version, 1.14.5, functioned without issues. The upgrade was performed following the standard procedure outlined in the documentation.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Server Side**
- Part-DB Version: [e.g. Part-DB 1.15.0]
- PHP Version: [e.g. PHP 8.1.2-1ubuntu2.20 ]
- Database Server [e.g. 10.6.18-MariaDB-0ubuntu0.22.04.1]

Cleared browser and server cache.
Verified configuration files and permissions.

There is no Information in the log files

jbtronics, 2025-01-07:
Have you build the frontend assets? Do any error occurs during the `yarn build` process?
Have you cleared the Part-DB cache after the building of the frontend dependencies (so that Part-DB recognizes the new pathes)?

trefase, 2025-01-07:
I am indeed serving Part-DB under a prefix(via reverse proxy) 

After pulling the latest commits it now works! Good work!

closed 2025-01-07 by trefase