Part-DB/Part-DB-server #825: Reverse proxy issue

Commiccannon, 2025-01-07:
**Describe the bug**
Reverse proxy was working untill i upgraded to 1.15. Same config now im getting a ERR_TIMED_OUT error, other services under the same reverse proxy are working fine

**To Reproduce**
Upgrade from 1.14 to 1.15.
Run a NGINX proxy under swag on same machine (10.0.0.21)
Have this in the docker-compose.yaml TRUSTED_PROXIES=127.0.0.1,127.0.0.0/8,::1,10.0.0.0/8,172.16.0.0/12,192.168.0.0/16  

**Expected behavior**
No change to proxy config during update, server responds as it was

**Server Side**
- Part-DB Version: 1.15.0
- PHP Version: 8.3.15
- Database Server sqlite 3.40.1

**Desktop (please complete the following information):**
 - Issue with every browser tired (chrome and edge) and both phone and PC

Commiccannon, 2025-01-07:
Partdb is running fine, I can access locally with no issues. The logs show no errors and no activity when I try and access externally

jbtronics, 2025-01-07:
How does your reverse proxy config look like?
Can your reverse proxy really reach part-DB? Does DNS resolving works?

I would expect that this is not really a Part-DB issue, but a an communication error between the reverse proxy and Part-DB.

Commiccannon, 2025-01-08:
Ive attached my reverse proxy config file.
Every device ive tried on my network can reach partdb and im having no issues with any other server under this reverse proxy.

I posted this issue becuase it happened as soon as i upgraded partdb. Have there been any changes to how the proxy needs to be set up at all?

Pytonballoon810, 2025-01-13:
I have the exact same setup as you do, but it works for me. The only difference is I address partdb by hostname and have it use the default port 80 since I am accessing it from within the container as both containers are in the same docker VLAN. Maybe try playing around with the port and double check that the IP is 100% the right one.