Part-DB/Part-DB-server #829: Clone part button not cloning

dominykasLi, 2025-01-10:
**Describe the bug**
I am adding SMD resistors en-masse. I clone the one resistor, rename it, then after saving and creating it, i clone the new part. The problem is, that sometimes, after clicking clone, changing the name and clicking save, it goes back to the part i was cloning. No new parts were created

**To Reproduce**

1. Go to (http://192.168.1.15/en/part/183/info#tools)
2. Click on Clone part
3. Change name from "1,1kR resistor", to "1,2kR resistor"
4. Click "Save changes".
5. Sometimes it creates the part, sometimes it just goes back to (http://192.168.1.15/en/part/183/info#tools)

**Expected behavior**
Always create a new part

**Screenshots**
[image]
When it works: 
[image]
When it does not, it just goes back to the original part. It does not say any message, especially, that it created a new part.

**Server Side**
- Part-DB Version: Version: 1.14.4
- PHP Version: one of the newest
- Database Server one of the newest
- I installed the server on my old laptop a month ago. I am using local Wifi to connect to the website.

**Desktop (please complete the following information):**
 - OS: Win10
 - Browser chrome
 - Version newest

dominykasLi, 2025-01-11:
Today is the same thing, sometimes does not create a new part. Had server Htop open, looks like it is not my server:
[image]

jbtronics, 2025-02-01:
This is might be some weird edge case regarding local history and Hotwire Turbo. The problem is that i cannot really reproduce it, making it nearly impossible to debug.

dominykasLi, 2025-02-02:
> This is might be some weird edge case regarding local history and Hotwire Turbo. The problem is that i cannot really reproduce it, making it nearly impossible to debug.

I did it like this:
1. Have a component open (like an SMD resistor).
2. Make a copy of it.
3. Change the name of the component and save it.
4. Open the new component.
5. Then clone it again (repeat steps for like 15 times, as fast as possible.)

closed 2025-05-13 by dominykasLi