Part-DB/Part-DB-server #832: 500 error on create part from provider

jsox79, 2025-01-13:
**Describe the bug**
500 error on create part from provider

**To Reproduce**
search for part in create part from provider
click the plus button

**Expected behavior**
the part to be added and confirmation of some sort

**Screenshots**

**Server Side**
- Part-DB Version: Part-DB Version | 1.15.1
- PHP Version: 8.3.15 (SAPI: fpm-fcgi)
- Database Server sqllite 3.40.1

**Desktop (please complete the following information):**
 - OS: any os
 - Browser any browser

jbtronics, 2025-01-14:
What does your server logs say about what the reason is for the 500?

I would probably assume that there is some issue with your API keys or some other configuration...

mkne, 2025-01-23:
Mentioning which information provider you are using might also be useful. I have come across a lot of this errors recently and often it is an error with the information providers API and not Part-DB.

Maystero14, 2025-02-04:
This issue Should've close. Problem was solved in v Part-DB Version | 1.15.2