Part-DB/Part-DB-server #835: Displaying "All" last activities fails with empty Error 500 message

mkne, 2025-01-14:
**Describe the bug**
Internal Server Error (Status 500) Error calling [/de/](http://172.16.0.56:8080/de/).
when trying to display "All" last activities. We have over 100 000 last activies.
No details are shown in the error message window, when clicking "View Details".
According to the PHP logs (see full logs in "Additional context") in Docker:
NOTICE: PHP message: PHP Fatal error:  Allowed memory size of 134217728 bytes exhausted (tried to allocate 20480 bytes) in /var/www/html/vendor/doctrine/orm/src/Internal/Hydration/ObjectHydrator.php on line 255
partdb  | - -  14/Jan/2025:06:47:23 +0000 "POST /index.php" 500
partdb  | NOTICE: PHP message: PHP Fatal error:  Allowed memory size of 134217728 bytes exhausted (tried to allocate 65536 bytes) in /var/www/html/vendor/composer/ClassLoader.php on line 576

**To Reproduce**
Steps to reproduce the behavior:
1. Open start page of part-db http://172.16.0.56:8080/de/
2. Change the number of Last activities to display from "10" to "All"
3. Wait during "Please Wait" is displayed
4. After some time the Error 500 is displayed

**Expected behavior**
Don't allow to display all "Last activities" if the memory consumption is to high. Or show a more useful error message explaining that the Memory size is exceeded.

Maybe limiting the amount of last activities in the configuration would also be useful. I havent't found a variable/parameter to limit the amount of log messages in the Configuration docs.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server Side**
- Part-DB Version: 1.15.1
- PHP Version: 8.3.15
- Database Server: sqlite

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36
 - Version 131.0.0.0

**Additional context**
partdb  | - -  14/Jan/2025:06:46:45 +0000 "GET /index.php" 200
partdb  | 172.16.1.75 - - [14/Jan/2025:06:46:45 +0000] "GET /de/ HTTP/1.1" 200 9317 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
partdb  | 172.16.1.75 - - [14/Jan/2025:06:46:46 +0000] "GET /de/tree/categories HTTP/1.1" 200 4044 "http://172.16.0.56:8080/de/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
partdb  | - -  14/Jan/2025:06:46:46 +0000 "GET /index.php" 200
partdb  | 172.16.1.75 - - [14/Jan/2025:06:46:46 +0000] "GET /de/tree/devices HTTP/1.1" 200 2777 "http://172.16.0.56:8080/de/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
partdb  | - -  14/Jan/2025:06:46:46 +0000 "GET /index.php" 200
partdb  | 172.16.1.75 - - [14/Jan/2025:06:46:46 +0000] "GET /de/tree/tools HTTP/1.1" 200 3939 "http://172.16.0.56:8080/de/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
partdb  | - -  14/Jan/2025:06:46:46 +0000 "GET /index.php" 200
partdb  | 172.16.1.75 - - [14/Jan/2025:06:46:46 +0000] "POST /de/ HTTP/1.1" 200 23011 "http://172.16.0.56:8080/de/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
partdb  | - -  14/Jan/2025:06:46:46 +0000 "POST /index.php" 200
partdb  | - -  14/Jan/2025:06:46:46 +0000 "POST /index.php" 204
partdb  | 172.16.1.75 - - [14/Jan/2025:06:46:46 +0000] "POST /csp/report HTTP/1.1" 204 914 "http://172.16.0.56:8080/de/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
partdb  | NOTICE: PHP message: PHP Fatal error:  Allowed memory size of 134217728 bytes exhausted (tried to allocate 20480 bytes) in /var/www/html/vendor/doctrine/orm/src/Internal/Hydration/ObjectHydrator.php on line 255
partdb  | - -  14/Jan/2025:06:47:23 +0000 "POST /index.php" 500
partdb  | NOTICE: PHP message: PHP Fatal error:  Allowed memory size of 134217728 bytes exhausted (tried to allocate 65536 bytes) in /var/www/html/vendor/composer/ClassLoader.php on line 576
partdb  | [Tue Jan 14 06:47:48.450087 2025] [proxy_fcgi:error] [pid 36:tid 60] [client 172.16.1.75:51463] AH01071: Got error 'PHP message: PHP Fatal error:  Allowed memory size of 134217728 bytes exhausted (tried to allocate 20480 bytes) in /var/www/html/vendor/doctrine/orm/src/Internal/Hydration/ObjectHydrator.php on line 255; PHP message: PHP Fatal error:  Allowed memory size of 134217728 bytes exhausted (tried to allocate 65536 bytes) in /var/www/html/vendor/composer/ClassLoader.php on line 576', referer: http://172.16.0.56:8080/de/
partdb  | 172.16.1.75 - - [14/Jan/2025:06:47:23 +0000] "POST /de/ HTTP/1.1" 500 267 "http://172.16.0.56:8080/de/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
partdb  | 172.16.1.75 - - [14/Jan/2025:06:47:24 +0000] "GET /de/log/107457/details HTTP/1.1" 200 8873 "http://172.16.0.56:8080/de/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
partdb  | - -  14/Jan/2025:06:47:24 +0000 "GET /index.php" 200

jbtronics, 2025-02-01:
There is not much that can be done to prevent this out of memory problem (because the history log gets thousands of entries very easily), and there cannot be much done as soon as it happens, so sending an reasonable message is also a bit difficult.

I have just disabled the "All" options for log tables, that should be fine for this case. Normally you don't want to view them all at once anyway.

closed 2025-02-01 by jbtronics

mkne, 2025-02-02:
> There is not much that can be done to prevent this out of memory problem (because the history log gets thousands of entries very easily), and there cannot be much done as soon as it happens, so [...]

That sounds good. I also told the user just not to use the option "All". Thanks for finding a quick way to fix it.

commit 3247a972 (2025-02-01): Removed "All" page length option for log tables, as this easily causes an out of memory situation on the server

You can easily have thousands of log entries, making it impossible to view them all at once.

This fixes #835