Part-DB/Part-DB-server #838: Information provider TME not working with private key instead of anonymous key

mkne, 2025-01-22:
**Describe the bug**
TME announced that the will drop support of anonymous keys from the end of February. Therefore I created a new private key and replaced the anonymous key in my docker-compose.yaml with the private key and restarted the container. Today my co-worker said that he got an error 500 when trying to add parts from TME. Yesterday my test seemed to be successful but maybe the part I searched was still in the cache? After switching back to the anonymous key it's working again. When creating the private key on the TME API webpage there are now also a customer number and a contact number mentioned. Maybe these have to be included in the API requests? Or the username and password? But we don't have such variables in the Part-DB configuration.

**Expected behavior**
Searching and adding parts from TME works with private keys.

**Server Side**
- Part-DB Version: 115.1
- PHP Version: 8.3.15
- Database Server: sqlite

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Chrome
 - Version: 132.0.6834.83

jbtronics, 2025-01-24:
What does the log says are the details for the 500 error? Are you sure its relateds to the anonymous key?

At least in the demo instance, the TME API works still fine: https://demo.part-db.de/en/tools/info_providers/search

mkne, 2025-01-25:
Sorry, I had to switch back to the anonymous key before I could check the logs and after restarting the container the logs were gone and I didn't want to block my colleagues with testing the private key again. Are you using a private key in the demo instance?

Maystero14, 2025-01-26:
Hi. I have a clean install of part-db. I want to add a new part to the database and during this process, I get an error 500

https://github.com/user-attachments/assets/3d52da25-2840-493b-a626-b491bee52fa9

I have new private key from TME. What am i doing wrong?

mkne, 2025-01-26:
@Maystero14 Your problem looks different than mine. In your case the API request works at least partly, because the part is found at TME. Maybe you can provide the detailled logs?

Maystero14, 2025-01-26:
It will be difficult to log, because I have Container with this volumes

[image]

mkne, 2025-01-27:
@Maystero14 I login on the machine running the docker container and change to the folder containing my docker-compose.yaml (cd /opt/part-db) and type: sudo docker compose logs -f 

I also don't have the logs mapped to a shared folder.

Maystero14, 2025-01-27:
I hope log file will help some...

[docker_deprecations.log](https://github.com/user-attachments/files/18562847/docker_deprecations.log)

jbtronics, 2025-01-27:
@Maystero14 these are just deprecation logs, that don't contain the relevant issue.

You need to look into the output of docker logs.

Maystero14, 2025-01-27:
Ok, I change it. Look now.

[docker.log](https://github.com/user-attachments/files/18563671/docker.log)

mkne, 2025-01-28:
OK, I have tried to create the part PCX2339J65105 from TME with the anonymous API key and with the private API key and also at my server I get an HTTP error 406 in the logs. What I have seen while switching between API keys. The private key is 5 characters longer than the ananymous key. Might this be an issue? However API access seems to function partly (it finds the part, but fails when I click on the +-Button). The exception with error 406 occurs when accessing https://api.tme.eu/Products/GetPricesAndStocks.json

[TME-Test-error406.txt](https://github.com/user-attachments/files/18569174/TME-Test-error406.txt)
[TME-Test-OK.txt](https://github.com/user-attachments/files/18569175/TME-Test-OK.txt)

mkne, 2025-01-28:
Maybe the issue is related to this information, that is displayed on the login page of the TME API website:
Changes in API security
We kindly inform that from 27 February 2017, the API service will be available only via the TLSv1.2 protocol.

We want to enhance your security while using the app, and the currently used libraries, i.e. SSLv3, TLSv1.0, TLSv1.1, are vulnerable to data capturing and deciphering.

More technical information can be found at
https://blog.cloudflare.com/yet-another-padding-oracle-in-openssl-cbc-ciphersuites/
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2016-2107

Maystero14, 2025-01-28:
> Maybe the issue is related to this information, that is displayed on the login page of the TME API website: Changes in API security We kindly inform that from 27 February 2017, the API service will [...]

So the question is, can this be fixed somehow, or is need to update?

jbtronics, 2025-01-28:
> We kindly inform that from 27 February 2017

The TLS1.2 has been the case since almost 8 years now, if that was really the problem, this should have appeared much earlier. Besides that Part-DB nowhere specifies what crypto algorithms to use. The http client (or more exactly the underlying curl), will just use the most secure crypto method available.

I would assume that the error 406 is caused by some changes in how the API works with the private keys, which will most likely need some changes in Part-DB. But i did not really had the time to look into this in more detailö

mkne, 2025-01-31:
I had a quick look into the repo from TME and there was a commit just 4 days ago (https://github.com/tme-dev/api-client-guzzle/pull/4/files) but this change was already implemented in https://github.com/Part-DB/Part-DB-server/blob/0f4238291f78a8f9fccc2e44b30cc3243a0b1850/src/Services/InfoProviderSystem/Providers/TMEClient.php#L62

But I found something that might be more relevant. The [API documentation](https://developers.tme.eu/documentation/download) from TME for the action Products/GetPricesAndStocks states:
GrossPrices : Boolean
Netto/gross price format switcher (**only if you are using anonymous token**). Optional parameter. Default "false"

Currently I have only set PROVIDER_TME_KEY & PROVIDER_TME_SECRET in my docker-compose.yaml

Is there a way to include the full http request/respond in the docker log files?

jbtronics, 2025-02-01:
> But I found something that might be more relevant. The [API documentation](https://developers.tme.eu/documentation/download) from TME for the action Products/GetPricesAndStocks states:
GrossPrices : Boolean
Netto/gross price format switcher (only if you are using anonymous token). Optional parameter. Default "false"

This was indeed the problem and the TME response even said that, when looking into it. As a fix you basically just have to set the `PROVIDER_TME_GET_GROSS_PRICES` env to 0.
I added an commit, which does this automatically if it detects that an private key is used.

Whether the prices include VAT or not is then determined by your account type with TME

closed 2025-02-01 by jbtronics