Part-DB/Part-DB-server #839: global_theme as environment variable via docker-compose.yaml

mkne, 2025-01-24:
**Is your feature request related to a problem? Please describe.**
I would love to set a default theme (for example "united") for new users other than the default bootstrap theme ("") via global_theme.
However it looks like there is no GLOBAL_THEME variable defined in the parameter.yaml/docker-compose.yaml?

**Describe the solution you'd like**
Enable setting a GLOBAL_THEME variable in docker-compose.yaml to one of the values from partdb.available_themes

**Describe alternatives you've considered**
Changing the default theme via the config/parameters.yaml file is not very useful when using docker compose, as it might/will be lost when updating the container. Also the documentation says: "You should expect, that you will have to do some manual merge, when you have changed something here and update to a newer version of Part-DB."

I haven't found any discussion or issue regarding this topic. I would also consider this very low priority. But it seems easy to implement.

jbtronics, 2025-01-24:
I am working on a new settings system, where you can configure things like this via the Web interface (you can still override them via env if you want that). The default theme will be one of the options.

So that should be configurable more easily in Part-DB 2.0.

mkne, 2025-01-31:
So, should I close this issue as it is already planned? @jbtronics The feature you've mentioned is already work in progress in the branch https://github.com/Part-DB/Part-DB-server/tree/settings-bundle, right?