Part-DB/Part-DB-server #840: Import parts with duplicate parameter or attachment names

mkne, 2025-01-28:
**Describe the bug**
I can not save a part imported from a service provider if the part has parameters with multiple parameter with the same name. For example, try to import U1242C from TME. I have to manually delete or rename the duplicate parameter name (for example: DC voltage measuring range).
The same problem happens with Attachments names (Image, Datasheet). But strangely I could import parts with duplicate attachment names but as soon as I change something and have to save the part again I get a error with the duplicate attachment names.

**Expected behavior**
Alternative 1:
Automatically import the parameters and allow duplicate Parameter names. (if possible with the DB schema, I'm not sure if parameter name is the primary key)

Alternative 2:
Automatically increment the Parameter name (for example: DC voltage measuring range 1, DC voltage measuring range 2, ...).

Alternative 3:
At least allow to bulk delete Parameters with duplicate names or delete all Parameters at once.

**Screenshots**

[image]

Server Side

Part-DB Version: 115.1
PHP Version: 8.3.15
Database Server: sqlite
Desktop (please complete the following information):

OS: Windows 11
Browser: Chrome
Version: 132.0.6834.83

jbtronics, 2025-02-01:
In principle, the database can store the duplicate parameter names, however this would lead to ambigious filtering using parameter constraints, and in my opinion a parameter should be a unique thing (as a part should not have two different value for the same property).

I have solved this (for now) that duplicate parameter names gets a numeric suffix, so that you get `Measurement`, `Measurement (2)`, `Measurement (3)` and so on. You probably still should review that and put it into a more usable parameters by hand, but this should simplify import process, if you dont care so much about these parameters.

closed 2025-02-01 by jbtronics

mkne, 2025-02-02:
I agree with you, I'm totally aware that this is not a good solution for filtering. I would be interested in how this is solved in the Information providers databases. Couldn't it be something like a multiple choice filter for a parameter? Like in my example for a multimeter if you're looking for a multimeter with "Measurement" of Voltage, Current, Capacitance, ... you can tick the wanted checkboxes. Of course you could also change the parameters into binary parameters like "Voltage measurement supported".