Part-DB/Part-DB-server #842: Symfony upgrade

darrieh, 2025-01-28:
Hello there, Part-DB community! I have a question regarding the Symfony version. I want to update the version to 7.2 as this is the next maintained version. I already tried so and faced a few issues ahead...
My Part-DB is installed via the docker image guide.

jbtronics, 2025-01-30:
The Symfony version will be upgraded to Symfony 7 with Part-DB 2.0 in the future.

It required at least PHP 8.2 so it will also change the requirements for Part-DB. Upgrading the Symfony version doesn't bring an immediate advantage to Part-DB, and the Symfony 6.4 will be supported as LTS for a while, therefore there is not too much of a hurry here I think...

darrieh, 2025-01-31:
Actually you are correct! I made a little further research and found out that the Symfony 6.4 is supported until 2027 so that is fine at the moment.

jbtronics, 2025-10-19:
Part-DB 2.0+ uses Symfony 7

closed 2025-10-19 by jbtronics