Part-DB/Part-DB-server #844: Changing the Port from 8080

jessecheu, 2025-01-30:
Hello,

Although port 8080 works, I wanted to change my port from 8080 to something like 5050.
From the instructions, I believe I just need to change the docker-compose.yaml but it seems changing the docker-compose.yaml under "ports" does not change the port and I am not able to access the web page. 
Is there somewhere else that I need to change to get it to work? 

My `docker-compose logs -f`:
```
partdb  | [30-Jan-2025 05:06:48] NOTICE: fpm is running, pid 23
partdb  | [30-Jan-2025 05:06:48] NOTICE: ready to handle connections
partdb  | [30-Jan-2025 05:06:48] NOTICE: systemd monitor interval set to 10000ms
partdb  | AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 192.168.16.2. Set the 'ServerName' directive globally to suppress this message
partdb  | AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 192.168.16.2. Set the 'ServerName' directive globally to suppress this message
partdb  | [Thu Jan 30 05:06:48.770120 2025] [mpm_event:notice] [pid 1:tid 1] AH00489: Apache/2.4.62 (Debian) configured -- resuming normal operations
partdb  | [Thu Jan 30 05:06:48.770240 2025] [core:notice] [pid 1:tid 1] AH00094: Command line: 'apache2 -D FOREGROUND'
```

jbtronics, 2025-01-30:
How does your docker-compose file look like?

LPTT, 2025-02-02:
changed mine to 8090, works fine.

services:
  partdb:
    container_name: partdb
    # By default Part-DB will be running under Port 8080, you can change it here
    ports:
      - '8090:80'
     ...