Part-DB/Part-DB-server #863: LCSC API Implementation

riacob, 2025-02-18:
**Is your feature request related to a problem? Please describe.**
LCSC Just released their API 

**Describe the solution you'd like**
LCSC provider to be updated for official API use

**Describe alternatives you've considered**
Keeping the current solution as a fallback

**Additional context**
[LCSC API Documentation](https://www.lcsc.com/docs/index.html)

jbtronics, 2025-02-18:
Interesting. I applied for an API key. Will see if and when they respond.
In general the API application asks for more (optional) infos than other distributors and seems to be intended for businesses. Lets see how easy it is to get an API key as a hobbyist.

d-buchmann, 2025-11-14:
I also applied for an API key (for a business).
This was the response I got after politely asking about progess after 5 months of silence:

> After reviewing your application, we regret to inform you that we’re unable to approve your API access at this time. It’s possible mainly due to the following reasons:
> - Your company type/size currently falls outside our API program’s priority access scope
> - We prioritize API access for partners with established order history and clear commercial procurement needs
> - Your application lacked sufficient details for us to assess compliance with our data security standards
>
> We understand this may disappoint you, but this decision reflects our current platform strategy—not a negative assessment of your business.

Even worse, they don't decline the application on the website, so you cannot reapply for API access with your account, correcting or updating the informatiuon that prevent them from accepting it.

I'm afraid we can't use this API anytime soon, at least as long as they run it in "priority access" mode, and the wording suggests they are barring hobbyists altogether.