Part-DB/Part-DB-server #866: Farnell/Element14 API Error

PressurisedAngus, 2025-02-21:
I've setup a partdb instance on a server and started connecting to the part distributor APIs. Mouser, LCSC, and TME works fine, however Farnell is proving to be an issue. I have created an application API with farnell's partner program and loaded in the key into the .env.local using uk.farnell.com.

When requesting data from farnell through info providers form, I get a status 500 error and the output to the log is: 
[partdblog.txt](https://github.com/user-attachments/files/18905275/partdblog.txt)

The request can be seen at [2025-02-21T10:12:29.055824+00:00].
 
PHP version: 8.2.26
Browser: Chrome V132.0.6834.197
OS: Windows 10 22H2
PartDB: V1.15.2

jbtronics, 2025-02-21:
I think the relevant issue here is "SSL certificate problem: unable to get local issuer certificate for"?
So it seems that the CA mouser uses to sign their TLS request, is not recognized as valid by your system.
So you should probably check tht your system CA storage is up to date.

Do you run Part-DB as server on Windows 10 22H2? Directly without docker?
Does your PHP has the curl extension available and usable. (Its not really a hard dependencies, but the non-curl methods to make http requests, are a bit odd sometimes...)

PressurisedAngus, 2025-02-21:
Apologies, i provided the incorrect information on the OS. It has been updated in the original post. Just to check,  the Mouser API works fine, however the farnell API is the issue. The PHP on the server has the latest version of curl, V1.88.1-10+deb12u8.

PressurisedAngus, 2025-02-21:
I ran update-ca-certificates which didn't find any that needed replaced. There aren't any in the list that obviously pertain to farnell. Is there a specific one i should check for?

RobThree, 2025-02-26:
I've been experiencing the same issue for a few weeks now. Initially I ignored it because I can use some other providers and thought it would be fixed sometime. I just upgraded to 1.16.0 and it was still not working.

I've run `apt update` and `update-ca-certificates`:

```bash
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
```

But I get the same error:

```bash
[2025-02-26T02:42:42.200746+01:00] cache.INFO: Lock acquired, now computing item "search_element14_1n4742a" {"key":"search_element14_1n4742a"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.X.X","http_method":"POST","server":"partdb.mydomain.local","referrer":"https://partdb.mydomain.local/en/tools/info_providers/search"}
[2025-02-26T02:42:42.201106+01:00] http_client.INFO: Request: "GET https://api.element14.com/catalog/products?term=any:1n4742a&storeInfo.id=nl.farnell.com&resultsSettings.offset=0&resultsSettings.numberOfResults=20&resultsSettings.responseGroup=large&callInfo.apiKey=******&callInfo.responseDataFormat=json&versionNumber=1.4" [] {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.X.X","http_method":"POST","server":"partdb.mydomain.local","referrer":"https://partdb.mydomain.local/en/tools/info_providers/search"}
[2025-02-26T02:42:42.261386+01:00] request.CRITICAL: Uncaught PHP Exception Symfony\Component\HttpClient\Exception\TransportException: "SSL certificate problem: unable to get local issuer certificate for "https://api.element14.com/catalog/products?term=any:1n4742a&storeInfo.id=nl.farnell.com&resultsSettings.offset=0&resultsSettings.numberOfResults=20&resultsSettings.responseGroup=large&callInfo.apiKey=******&callInfo.responseDataFormat=json&versionNumber=1.4"." at ErrorChunk.php line 56 {"exception":"[object] (Symfony\\Component\\HttpClient\\Exception\\TransportException(code: 0): SSL certificate problem: unable to get local issuer certificate for \"https://api.element14.com/catalog/products?term=any:1n4742a&storeInfo.id=nl.farnell.com&resultsSettings.offset=0&resultsSettings.numberOfResults=20&resultsSettings.responseGroup=large&callInfo.apiKey=******&callInfo.responseDataFormat=json&versionNumber=1.4\". at /var/www/partdb/vendor/symfony/http-client/Chunk/ErrorChunk.php:56)\n[previous exception] [object] (Symfony\\Component\\HttpClient\\Exception\\TransportException(code: 0): SSL certificate problem: unable to get local issuer certificate for \"https://api.element14.com/catalog/products?term=any:1n4742a&storeInfo.id=nl.farnell.com&resultsSettings.offset=0&resultsSettings.numberOfResults=20&resultsSettings.responseGroup=large&callInfo.apiKey=******&callInfo.responseDataFormat=json&versionNumber=1.4\". at /var/www/partdb/vendor/symfony/http-client/Response/CurlResponse.php:324)"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.X.X","http_method":"POST","server":"partdb.mydomain.local","referrer":"https://partdb.mydomain.local/en/tools/info_providers/search"}
[2025-02-26T02:42:42.266786+01:00] deprecation.INFO: User Deprecated: Since nelmio/security-bundle 3.4.0: The "Nelmio\SecurityBundle\EventListener\XssProtectionListener" class is deprecated, use Content Security Policy without allowing "unsafe-inline" scripts instead. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since nelmio/security-bundle 3.4.0: The \"Nelmio\\SecurityBundle\\EventListener\\XssProtectionListener\" class is deprecated, use Content Security Policy without allowing \"unsafe-inline\" scripts instead. at /var/www/partdb/vendor/nelmio/security-bundle/src/EventListener/XssProtectionListener.php:35)"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.X.X","http_method":"POST","server":"partdb.mydomain.local","referrer":"https://partdb.mydomain.local/en/tools/info_providers/search"}
```

The error above:
`SSL certificate problem: unable to get local issuer certificate for "https://api.element14.com/catalog/products?term=any:1n4742a&storeInfo.id=nl.farnell.com&resultsSettings.offset=0&resultsSettings.numberOfResults=20&resultsSettings.responseGroup=large&callInfo.apiKey=******&callInfo.responseDataFormat=json&versionNumber=1.4"`

When I visit [the URL](https://api.element14.com/catalog/products?term=any:1n4742a&storeInfo.id=nl.farnell.com&resultsSettings.offset=0&resultsSettings.numberOfResults=20&resultsSettings.responseGroup=large&callInfo.apiKey=****&callInfo.responseDataFormat=json&versionNumber=1.4`)* on a Windows machine in my browser it works fine.
<sup>\* API Key obscured</sup>

When I run this in my PartDB VM: ` wget https://api.element14.com/catalog/products\?term...` I get:

```bash
Resolving api.element14.com (api.element14.com)... 34.249.183.152, 52.31.148.70, 52.213.192.196
Connecting to api.element14.com (api.element14.com)|34.249.183.152|:443... connected.
ERROR: The certificate of ‘api.element14.com’ is not trusted.
ERROR: The certificate of ‘api.element14.com’ doesn't have a known issuer.
```

So there is, indeed, an issue with the issuer(s). I'm not quite sure how to go from here.

OS: Debian 12

mkne, 2025-03-12:
@RobThree I've tried your solution of manually adding the certificates. And update-ca-certificates seemed to have added the certificates but if I run:
```
grep -r "SSL.com TLS RSA Root CA 2022" /etc/ssl/certs/
grep -r "Entrust OV TLS Issuing RSA CA 1" /etc/ssl/certs/
```
it still can't find any files and 
`openssl s_client -connect api.element14.com:443 -servername api.element14.com -showcerts`
still fails with a 400 Bad Request saying 
Verify return code: 2 (unable to get issuer certificate)

jbtronics, 2025-03-13:
@RobThree Could you find out, if there is a reason that this certificate is missing in the debian certificate store?

RobThree, 2025-03-13:
> [@RobThree](https://github.com/RobThree) Could you find out, if there is a reason that this certificate is missing in the debian certificate store?

I'm afraid I wouldn't know where to start 😅 What I can, and will, do is do a quick clean installation of Debian, update it all the way, and see if they're (still) missing.

closed 2025-03-27 by jbtronics