Part-DB/Part-DB-server #869: Bulk import from info provider

riacob, 2025-02-23:
**Is your feature request related to a problem? Please describe.**
I want to transfer 170+ parts from an excel sheet to parts-db. 
Importing it with the normal import functionality, regardless if the spn# is present or not, does not automatically consider the import to be "from info provider", thus requiring manual copy, paste, and stock setting for every single part.

**Describe the solution you'd like**
An extra import option, be it a separate menu item or a checkmark, to import CSV files containing spn#, stock, and category, and automatically fetching data for each of them from the selected info provider.

**Describe alternatives you've considered**
/

**Additional context**
In my particular situation, i need to move from a google sheet, which has all my parts in it, to parts-db. 
And, in the future, be able to seamlessly import order history from LCSC to part-db, as I'm currently manually copying the mpn# into the google sheet, then using IMPORTXML magic to fetch the data, which is very inefficient.

riacob, 2025-02-23:
Proposed solution is a for loop calling the import from info provider function with the mpn# as param, then setting category and stock from csv, and fetching the data. Repeating this n times.

riacob, 2025-02-23:
If you wish, I can fork this and try to build a solution. However, I need some info on contributing and installing the build and debug system you're using.

barisgit, 2025-08-01:
First step i envision as this: User selects N parts with select and maps which fields they want to map where (eg. mpn -> LCSC and to MOUSER, LCSC spn to LCSC, Mouser spn to Mouser, maybe even skip normal mpn...). Then the API in the background fetches everything and adds to cache. The import is then very fast just clicking buttons.

barisgit, 2025-08-02:
#982 

Here, we also needed that, I think it is pretty neat will test it next week in real circumstances.

barisgit, 2025-10-01:
#1060 Is it connected to this issue?

Nick-C130, 2025-10-01:
No its not that, i have recently made the oauth and its working fine while adding single components
Thanks

RawSmokeTerribilus, 2025-11-16:
Hello awesome team! I have been trying to make it work but it's not lol. I got:

`[2025-11-16T11:36:59.057031+00:00] request.CRITICAL: Uncaught PHP Exception Error: "Object of class App\Services\InfoProviderSystem\Providers\LCSCProvider could not be converted to string" at BulkInfoProviderService.php line 192 {"exception":"[object] (Error(code: 0): Object of class App\\Services\\InfoProviderSystem\\Providers\\LCSCProvider could not be converted to string at /var/www/html/src/Services/InfoProviderSystem/BulkInfoProviderService.php:192)
[stacktrace]
#0 /var/www/html/src/Services/InfoProviderSystem/BulkInfoProviderService.php(192): array_intersect()
#1 /var/www/html/src/Services/InfoProviderSystem/BulkInfoProviderService.php(78): App\\Services\\InfoProviderSystem\\BulkInfoProviderService->processRegularProviders()
#2 /var/www/html/src/Controller/BulkInfoProviderImportController.php(216): App\\Services\\InfoProviderSystem\\BulkInfoProviderService->performBulkSearch()
#3 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(183): App\\Controller\\BulkInfoProviderImportController->step1()
#4 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
#5 /var/www/html/vendor/symfony/http-kernel/Kernel.php(182): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#6 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#7 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKerne
lRunner->run()
#8 /var/www/html/public/index.php(5): require_once('...')
#9 {main}
"}`
There is nothing my vibe coding skills can do lol. I am running the last docker image, readed all the issues, pull requests and wiki. tried all the possible combinations, then spent a couple days checking if it was my fault, it is not. It's broken.
I add my stock with the import option, getting all the info from LCSC (i use their cart csv). That means that i have thousands of parts that lack info/attachements, when i update them one by one, everything goes flawlessly. But, impossible for me to do that with all the parts so, here i am waiting like a kid to see if we can get it sorted lol. I would love too if someone points me in the good direction, i may be doing something wrong? (tried every possibility 5 times buit, who knows)

RawSmokeTerribilus, 2025-11-16:
> 

> Hello awesome team! I have been trying to make it work but it's not lol. I got:
>
> `[2025-11-16T11:36:59.057031+00:00] request.CRITICAL: Uncaught PHP Exception Error: "Object of class [...]

HOLLY COW... I vibe coded it lol... going to fork, copy paste the file that i worked on and create a pull request. working on docker perfectly. don´t hate me, i am not going to do this anymore, it's just that it is such a fantastic feature and it wasn´t working. love you guys, vibe coding sucks, don't do it.