Part-DB/Part-DB-server #872: Encountered 500 error while searching for creating parts from info provider LCSC. Version1.16.0

vrxiaojie, 2025-02-25:
**Describe the bug**
I upgraded partDB from 1.15.2 to 1.16, and other pages are working properly. I only encountered a 500 error when creating parts from the info provider LCSC. Downgrade the version to 1.15.1, and this feature works normally as before.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade docker image to 1.16.0
2. Run `docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate` . Migration Successful.
3. Visit partDB in browser and open the URL `http://192.168.50.111:8000/en/tools/info_providers/search` 
4. Typing the keyword `C2907078` (Supplier part number) and choosing the Provider `LCSC`

[image]

5. The 500 error pops up.

[image]

6. Try running `php bin/console cache:clear` and it doesn't help. **Even worse**, every page displays 500 error.

**Expected behavior**
Display the correct results of my search. (this screenshot captured in version 1.15.1)

[image]

**Server Side**
- Synology DSM 7.2.2 docker
- Part-DB Version: partdborg/part-db:1.16.0  (running in docker)
- Database Server MySQL 8.3.0 (running in docker)

**Desktop :**
 - OS: Windows 11
 - Browser: Edge
 - Version: 133.0.3065.82

Logs: 
1. This log records what appeared after I searched for parts from info provider:

[info_provider_search_log.txt](https://github.com/user-attachments/files/18952293/info_provider_search_log.txt)

2. This log records what happened after I tried to clear the cache:
[after_clear_cache_log.txt](https://github.com/user-attachments/files/18952306/after_clear_cache_log.txt)

mkne, 2025-02-25:
I think it's this change: https://github.com/Part-DB/Part-DB-server/commit/80527e35c39c49cba0f8bef88ca9b613af3ab61b#diff-1a62de89ea90e259023b016f9d0394efa0dcbdd29b3ed187627a587163b1f185

I'll fix that.

vrxiaojie, 2025-02-25:
> I think it's this change: [80527e3#diff-1a62de89ea90e259023b016f9d0394efa0dcbdd29b3ed187627a587163b1f185](https://github.com/Part-DB/Part-DB-server/commit/80527e35c39c49cba0f8bef88ca9b613af3ab61b#diff-1a62de89ea90e259023b016f9d0394efa0dcbdd29b3ed187627a587163b1f185)
> 
> I'll fix that.

Thank you ! It works fine.

closed 2025-02-25 by vrxiaojie