Part-DB/Part-DB-server #876: Problem with update on DOCKER from 1.15.1 to 1.16.1

Maystero14, 2025-03-03:
I have PART-DB version 1.15.1 running in a Docker Container Station on my QNAP. I decided to update it to the latest version, 1.16.1.

To do this, I ran the following command:
`docker pull jbtronics/part-db1:latest`

[image]

Container Station downloaded the latest image (1.16.1), but after restarting the container, nothing changed. The PART-DB version remains the same as before the update.

[image]

At first, I thought I had broken something and asked my colleague to check the update on his QNAP Container Station. After testing, he got the same result. This raises the question: are we doing something wrong? Or is it some kind of issue with the Docker images?

jbtronics, 2025-03-07:
Have you somehow mapped the var/cache folder of part-DB to a folder or a volume?
Are you sure that you are acutally seeing the output of the correct docker container?

The docker images, itself should be fine.

Maystero14, 2025-03-07:
Hello Jan.
He is 100% sure about his volume configuration in Docker. I have done many tests. I even ran a test version of Docker to find the cause of the problem.  

So, I first ran Docker in the test version with the Part-DB 1.5.1 image and tried to upgrade it to version 1.15.1.

`docker pull jbtronics/part-db1:v1.15.2`

Unfortunately, the automatic update of the image to 1.5.2 had no effect. The image was downloaded correctly, and Docker itself started without any issues after the update. However, Part-DB still displayed version 1.5.1.  

I searched for a solution for a long time. In the end, I was only able to fix the problem when I manually changed the directory name.

`/Container/container-station-data/lib/docker/volumes/part-db-test_html/_data`

and created a new empty directory
`_data`

Then, I executed the "Recreate" command in Docker. After restarting Docker, the Part-DB version was successfully updated to the newer version. It's not the most elegant solution, but it works.  

This way, I managed to update Part-DB step by step to the latest version, 1.16.1.

[image]

I’m not an expert in creating Docker images, but it seems to me that this is some kind of issue with the image itself. I have conducted many tests and also asked a colleague to run tests in his test environment. He encountered exactly the same problems as I did.

If needed, I can test Docker images for you 😊

closed 2026-08-16 by jbtronics