Part-DB/Part-DB-server #882: DB Migration error after going from 1.15.2 to 1.16.0

aladaris, 2025-03-06:
**Describe the bug**
After updating to version 1.16.0 and executing the documented command `sudo -u www-data php bin/console doctrine:migrations:migrate` I get an error whenever I login or click on any category.

If I load into the application without doing any login I can browse some sections. Like everything in Tools. As for the Edit section, I can go into each of the items (Attachment Types, Categories, Projects, ...), but if I click on any of the existing items I get a 500 http error. Samething happens if I click on any category on top of the left menu.

If I perform a login I get an http 500 error just by loading the home page.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the home page without doing any login
2. Click on any category
4. See 

> error: 500
Internal Server Error
Something bad happened internally.
There is nothing you could do about this, except trying to reload the page.
You can try to [Go Back](javascript:history.back()) or Visit the homepage.
If this error persists, please contact your administrator.
Infos for Admin: Error while executing database query.
This is maybe caused by an old database schema.

**Expected behavior**
I shouldn't see a page displaying a http 500 error

**Screenshots**
[error 500 page](https://imgur.com/a/KI2WUky)

**Server Side**
- Docker version
- Part-DB Version: jbtronics/part-db1:1.16.0
- Database Server sqlite

**Additional context**
Log file describing the container being created, an unauthenticated user going into the home page and then clicking on one of the categories on the left menu: [log 1](https://pastebin.com/N68pp81E)

Log file describing the container being created, an unauthenticated user going into the home page and then going into the login page and performing a login action: [log 2](https://pastebin.com/1USZAVZb)

This is the compose file I use to execute the service: [compose.yaml](https://pastebin.com/E4GFHgAU)

jbtronics, 2025-03-07:
That is most likely the same problem as in #871. You must run the migrations with `sudo docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate`, or if you want to to use the sudo command inside the docker container, you must use `sudo -E`. Otherwise the console command does not see the env variables containing the database path, and applies the migrations to the wrong database...

aladaris, 2025-03-09:
Oh this has worked just fine.
The `sudo -E` hint totally did it for me. I saw some warnings during the previous execution, but just ignored them.
Preserving the environment context makes toal sense :)

Thank you!

closed 2025-03-09 by aladaris