Part-DB/Part-DB-server #885: No parts returned from the KiCAD intigration API

Joe-User, 2025-03-07:
**Describe the bug**
I am attempting to use the EDA integration feature with KiCAD 9.0 and no parts are returned.

**To Reproduce**
1. Setup a read-only API key for my user on Part-DB
2. Added some info to 2 parts on the "EDA Information" tab and checked the "Force visibility" box
3. Created a file `partd.kicad_httplib` file using this template from the documentation:`{
 "meta": {
     "version": 1.0
 },
 "name": "Part-DB library",
 "description": "This KiCAD library fetches information externally from ",
 "source": {
     "type": "REST_API",
     "api_version": "v1",
     "root_url": "my.server.local/en/kicad-api/",
     "token": "THE_GENERATED_API_TOKEN"
 }
}    
`
3. Attempted to find the parts in the KiCAD listing and nothing was found.
4. Ran an HTTP\GET request including my API key in the header as a bearer token on https://my.server.local/en/kicad-api/v1 and received 200 OK and the following response:
`{
    "categories": "",
    "parts": ""
}`

**Expected behavior**
Expecting to see a listing of my parts in the KiCAD libraries or an API return with parts listed.

**Server Side**
- Part-DB Version: 1.15.1
- PHP Version: 8.3.15
- Database Server: 8.0.41

**Desktop:**
 - OS: Windows 11 Pro
 - Browser:  Chrome Version 133.0.6943.143

**Additional context**
I have also attempted a restart of the service and ran the `sudo -u www-data php bin/console cache:clear` command from looking at some older posts.

jbtronics, 2025-03-29:
> 5\. Ran an HTTP\GET request including my API key in the header as a bearer token on https://my.server.local/en/kicad-api/v1 and received 200 OK and the following response:
>     `{ "categories": "", "parts": "" }`

That is normal behavior, the values there has no meaning for KiCAD (see https://dev-docs.kicad.org/en/apis-and-binding/http-libraries/index.html).

The relevant endpoint is `en/kicad-api/v1/categories.json`, that should show the list of available categories.

jbtronics, 2025-03-29:
There was a caching bug, that a category does not necessarily become visible, after the part was changed so it becomes visible. This is fixed now.
However this should have been updated, when you cleared the Part-DB cache.

Have you restarted KiCAD? It caches the results and doesnt notice changes to Part-DB until you restart KiCAD.

closed 2026-02-25 by jbtronics