Part-DB/Part-DB-server #887: attachments sorted by type of attachment or tree view

majojoe, 2025-03-09:
It would be nice if it was possible to sort the attachments according the attachment type.
As an alternative it would be nice to have a tree view of the attachments based on the various attachment types. 
This would help a lot to keep the view clear if there are a lot of attachments (datasheets, app notes, certificates, PCN's/PDN's etc.).

d-buchmann, 2025-05-20:
Have you seen Tools -> Show -> Attachments at the bottom of the side bar?
You can use sophisticated filters and sorting if you like.

majojoe, 2025-05-27:
Yes, thanks, I have seen that option. But, if I am working with a part then there can be quite a lot of documents for a part (atasheets, app notes, certificates, PCN's/PDN's etc.). Sometimes it can be quite take a moment to find the correct document type. A sort function based on the attachment type or a tree view could help here to sort things a little better.

d-buchmann, 2025-05-28:
Now I get it - you're talking about the part info view.
This is set in `templates/parts/info/_attachments_info.html.twig` in case you/anyone wants to play with it and put up a PR.
I'm afraid I won't have much time to look into this.