Part-DB/Part-DB-server #888: "500 Internal Server Error" when creating a new part from provider (Mouser)

romihsan, 2025-03-10:
**Describe the bug**
I am able to search Mouser for parts and I am able to create new parts in my DB by clicking on the + icon on the right hand side, but when I want to add one particular part number (KGL21DR71E104KU), it throws the 500 error.

The following image shows the search result:
[image]

What I found is that when trying to add the top part to the DB, I get the 500 error, but if I click the + button for the 2nd part it loads the create part dialog as usual.

**Server Log**
I have copied all the log entries outputted when I click the + button:
```
[2025-03-10T00:21:15.679931+00:00] request.INFO: Matched route "info_providers_create_part". {"route":"info_providers_create_part","route_parameters":{"_route":"info_providers_create_part","_controller":"App\\Controller\\PartController::createFromInfoProvider","_locale":"en","providerKey":"mouser","providerId":"581-KGL21DR71E104KU"},"request_uri":"http://192.168.1.172:8080/en/part/from_info_provider/mouser/581-KGL21DR71E104KU/create","method":"GET"} {"token":null,"url":"/en/part/from_info_provider/mouser/581-KGL21DR71E104KU/create","ip":"192.168.1.77","http_method":"GET","server":"192.168.1.172","referrer":"http://192.168.1.172:8080/en/tools/info_providers/search"}
[2025-03-10T00:21:15.680976+00:00] deprecation.INFO: User Deprecated: Since api-platform/symfony 4.1: The "api_platform.validator.query_parameter_validator" service is deprecated use "\ApiPlatform\Metadata\Parameter::$constraints" instead. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since api-platform/symfony 4.1: The \"api_platform.validator.query_parameter_validator\" service is deprecated use \"\\ApiPlatform\\Metadata\\Parameter::$constraints\" instead. at /var/www/html/var/cache/docker/ContainerGlRTxxV/getApiPlatform_Validator_QueryParameterValidatorService.php:23)
[stacktrace]
#0 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(201): ContainerGlRTxxV\\getApiPlatform_Validator_QueryParameterValidatorService::do()
#1 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(1306): ContainerGlRTxxV\\App_KernelDockerContainer->load()
#2 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(400): ContainerGlRTxxV\\App_KernelDockerContainer::getApiPlatform_Listener_View_ValidateQueryParametersService()
#3 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(257): ContainerGlRTxxV\\App_KernelDockerContainer::ContainerGlRTxxV\\{closure}()
#4 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
#5 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
#6 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(157): Symfony\\Component\\EventDispatcher\\EventDispatcher->dispatch()
#7 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
#8 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#9 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#10 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#11 /var/www/html/public/index.php(5): require_once()
#12 {main}
"} {"token":null,"url":"/en/part/from_info_provider/mouser/581-KGL21DR71E104KU/create","ip":"192.168.1.77","http_method":"GET","server":"192.168.1.172","referrer":"http://192.168.1.172:8080/en/tools/info_providers/search"}
[2025-03-10T00:21:15.681190+00:00] deprecation.INFO: User Deprecated: Since api-platform/core 3.4: The class "ApiPlatform\ParameterValidator\ParameterValidator" is deprecated, use "\ApiPlatform\Metadata\Parameter::$constraints" instead. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since api-platform/core 3.4: The class \"ApiPlatform\\ParameterValidator\\ParameterValidator\" is deprecated, use \"\\ApiPlatform\\Metadata\\Parameter::$constraints\" instead. at /var/www/html/vendor/api-platform/core/src/ParameterValidator/ParameterValidator.php:41)
[stacktrace]
#0 /var/www/html/var/cache/docker/ContainerGlRTxxV/getApiPlatform_Validator_QueryParameterValidatorService.php(31): ApiPlatform\\ParameterValidator\\ParameterValidator->__construct()
#1 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(201): ContainerGlRTxxV\\getApiPlatform_Validator_QueryParameterValidatorService::do()
#2 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(1306): ContainerGlRTxxV\\App_KernelDockerContainer->load()
#3 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(400): ContainerGlRTxxV\\App_KernelDockerContainer::getApiPlatform_Listener_View_ValidateQueryParametersService()
#4 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(257): ContainerGlRTxxV\\App_KernelDockerContainer::ContainerGlRTxxV\\{closure}()
#5 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
#6 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
#7 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(157): Symfony\\Component\\EventDispatcher\\EventDispatcher->dispatch()
#8 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
#9 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#10 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#11 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#12 /var/www/html/public/index.php(5): require_once()
#13 {main}
"} {"token":null,"url":"/en/part/from_info_provider/mouser/581-KGL21DR71E104KU/create","ip":"192.168.1.77","http_method":"GET","server":"192.168.1.172","referrer":"http://192.168.1.172:8080/en/tools/info_providers/search"}
[2025-03-10T00:21:15.681279+00:00] deprecation.INFO: User Deprecated: Since api-platform/core 3.4: The class "ApiPlatform\ParameterValidator\Validator\ArrayItems" is deprecated, use "\ApiPlatform\Metadata\Parameter::$constraints" instead. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since api-platform/core 3.4: The class \"ApiPlatform\\ParameterValidator\\Validator\\ArrayItems\" is deprecated, use \"\\ApiPlatform\\Metadata\\Parameter::$constraints\" instead. at /var/www/html/vendor/api-platform/core/src/ParameterValidator/Validator/ArrayItems.php:25)
[stacktrace]
#0 /var/www/html/vendor/api-platform/core/src/ParameterValidator/ParameterValidator.php(46): ApiPlatform\\ParameterValidator\\Validator\\ArrayItems->__construct()
#1 /var/www/html/var/cache/docker/ContainerGlRTxxV/getApiPlatform_Validator_QueryParameterValidatorService.php(31): ApiPlatform\\ParameterValidator\\ParameterValidator->__construct()
#2 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(201): ContainerGlRTxxV\\getApiPlatform_Validator_QueryParameterValidatorService::do()
#3 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(1306): ContainerGlRTxxV\\App_KernelDockerContainer->load()
#4 /var/www/html/var/cache/docker/ContainerGlRTxxV/App_KernelDockerContainer.php(400): ContainerGlRTxxV\\App_KernelDockerContainer::getApiPlatform_Listener_View_ValidateQueryParametersService()
#5 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(257): ContainerGlRTxxV\\App_KernelDockerContainer::ContainerGlRTxxV\\{closure}()
#6 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
#7 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
#8 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(157): Symfony\\Component\\EventDispatcher\\EventDispatcher->dispatch()
[... cut ...]
192.168.1.77 - - [10/Mar/2025:00:21:15 +0000] "GET /en/part/from_info_provider/mouser/581-KGL21DR71E104KU/create HTTP/1.1" 500 4367 "http://192.168.1.172:8080/en/tools/info_providers/search" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/134.0.0.0 Safari/537.36"
- -  10/Mar/2025:00:21:15 +0000 "GET /index.php" 500
```

I'm not sure what the issue could be.
I've also updated to 1.16.1 (from 1.15.?) after running into this issue, but it still persisted.

Cheers!

P.S. Apologies for including all those logs, but I wanted to be sure I included everything that happened after clicking the + button.

crackn, 2025-03-14:
same error cloning from octopart provider.

jbtronics, 2025-03-27:
This was basically issue #616, however i found a better solution, to properly fix this with the latest commit.

closed 2025-03-27 by jbtronics