Part-DB/Part-DB-server #891: Element14 API not working in Docker image, likely due to untrusted certificate

turmoni, 2025-03-11:
**Describe the bug**
I'm pretty sure this is basically https://github.com/Part-DB/Part-DB-server/issues/866 , but specifically raising it as an issue with the (non-frankenphp) container image.

**To Reproduce**
Steps to reproduce the behavior:
1. Install and run jbtronics/part-db1 with an element14 configuration set up
2. Try to do something with this integration (e.g. creating a part from the info provider)
3. Observe the 500 error, and the "SSL certificate problem: unable to get local issuer certificate for [...]" error in the logs

**Expected behavior**
The integration should work

**Screenshots**
Not needed

**Server Side**
- Part-DB Version: container image jbtronics/part-db1:latest with digest ID sha256:2b96a4714babda0cabf4a35db70679a758f91593d7680c338384e56bcbb95035
- PHP Version: As in the above image
- Database Server: SQLite

**Desktop (please complete the following information):**
N/A
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
N/A
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
I said it's the non-frankenphp image, but I don't know if that's also affected, I just don't run that one.

jbtronics, 2025-03-13:
Yeah the docker image uses debian, so it will have the same issue with the missing certificate as a bare debian installation.
The frankenphp image is probably affected the same way.

jbtronics, 2025-03-27:
Should be fixed with latest commit.
Part-DB now uses a separate trusted CA file, that contains the required cert, no matter what the system says.

closed 2025-03-27 by jbtronics