Part-DB/Part-DB-server #896: Another 500 error, AGAIN

sochartgit, 2025-03-21:
I am so fuc... tired of this instable piece of software.
Everything was running fine and then Bang, error 500 again.
I am only trying to add a part by clicking the + button.
It was working 5 minutes ago with the same bloody settings, same server, same stuff.
It's so unstable I am thinking of getting rid of that and coming back to Excel, seriously.

[image]

[image]

sochartgit, 2025-03-21:
Version: 1.16.1

virtadpt, 2025-03-21:
The output page says "check var/log/prod.log for additional information."  Have you done so?  What's in there?

cnf, 2025-03-21:
Yeah, i am getting random 500's. mostly it's because images are missing? i have a category that insists it has an item in it, but i can't see any item. I seem to be randomly losing data at places...

not sure what is going on...

sochartgit, 2025-03-21:
You can search for CM6-225 from Mouser
Nope, these are almost the same item from the same seller (Mouser) with pictures and descriptions.
2 first are giving error 500, third one is working.

[image]

sochartgit, 2025-03-21:
This is the error in the logs:
172.18.0.20 - - [21/Mar/2025:22:49:50 +0000] "GET /en/part/from_info_provider/mouser/616-63074-510-039/create HTTP/1.1" 500 6782 "https://XXXXXXXXXXXXXXXX/en/tools/info_providers/search" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:136.0) Gecko/20100101 Firefox/136.0"

jbtronics, 2025-03-29:
Thats most likely related to issue #888 and #616, which should now be fixed.

closed 2025-09-07 by jbtronics