Part-DB/Part-DB-server #898: Random data loss, i guess?

cnf, 2025-03-23:
**Describe the bug**
I randomly get parts / categories etc that show error 500 when trying to view them.

For parts, it's often the attachments? manually going to `/edit` and selecting `Download external file` and saving seems to solve the issue

I now have a category I can not acces. not view nor edit. I have no clue how to solve this...

These are all parts and categories that worked fine, at some point.

**To Reproduce**
Steps to reproduce the behavior:
Uhm, I have no idea, it _seems_ random to me.

**Expected behavior**
Not lose data :P

**Server Side**
- Part-DB Version: [e.g. Part-DB 1.0 (for dev versions commit hash is useful]
```
Part-DB Version 	1.16.1
Symfony environment 	docker (Debug: False )
Part-DB Instance name 	Part-DB
Default locale 	English (en)
Default timezone 	Europe/Berlin
Default Currency 	Euro (EUR, €)
Default theme 	bootstrap
Enabled locales 	English (en) German (de) Italian (it) French (fr) Russian (ru) Japanese (ja) Czech (cs) Danish (da) Chinese (zh) Polish (pl)
Demo Mode 	False
GDPR Compliance Mode 	True
Users
Use Gravatar for Avatar 	True
Password Reset via Email enabled 	False
Single Sign-On via SAML enabled 	False
Email
Configured E-Mail sender 	noreply@partdb.changeme (Part-DB Mailer)
Attachments
Allow server-side download of attachments 	True
Detailed error pages enabled 	True (Admin Contact email: )
Maximum file size for attachments 	100M (Effective: 100.00 MB)
```

- PHP Version: [e.g. PHP 7.4.3]
```
PHP version 	8.3.17 (SAPI: fpm-fcgi)
Server Operating System 	Linux bacabda6-partdb 6.12.18-haos #1 SMP PREEMPT_DYNAMIC Mon Mar 17 11:16:31 UTC 2025 x86_64
PHP Bit Size 	64-bit
Opcache enabled 	True
Maximum upload size (upload_max_filesize / post_max_size) 	256M / 300M
PHP extensions 	Core date libxml openssl pcre zlib filter hash json random Reflection SPL session standard sodium cgi-fcgi mysqlnd PDO xml bcmath calendar ctype curl dom mbstring FFI fileinfo ftp gd gettext iconv intl exif mysqli pdo_mysql pdo_pgsql pdo_sqlite pgsql Phar posix readline shmop SimpleXML sockets sqlite3 sysvmsg sysvsem sysvshm tokenizer xmlreader xmlwriter xsl zip Zend OPcache
Server time 	March 23, 2025 at 6:14:22 PM GMT+1
Symfony Kernel Runtime 	Symfony\Component\Runtime\SymfonyRuntime
Symfony Kernel Runtime Environment 	docker
Symfony Kernel Runtime mode 	web: 1
```
- Database Server [e.g. MySQL 5.7]
```
Database type 	sqlite
Database Server Version 	3.40.1
Database Size 	1.08 MB
Database Name 	main
Database User 	sqlite
Natural sort method 	Disabled
Slow natural sort allowed 	False 
```

**Desktop (please complete the following information):**

Linux firefox

**Additional context**
[... 280 lines cut ...]
#8 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
#9 /var/www/html/vendor/symfony/http-kernel/EventListener/ErrorListener.php(117): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#10 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(260): Symfony\\Component\\HttpKernel\\EventListener\\ErrorListener->onKernelException()
#11 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
#12 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
#13 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(239): Symfony\\Component\\EventDispatcher\\EventDispatcher->dispatch()
#14 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(91): Symfony\\Component\\HttpKernel\\HttpKernel->handleThrowable()
#15 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#16 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#17 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#18 /var/www/html/public/index.php(5): require_once()
#19 {main}
"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"cnf"},"url":"/en/category/11/parts","ip":"100.91.100.100","http_method":"GET","server":"parts.mydomain.com","referrer":null}
[2025-03-23T17:10:25.430847+00:00] security.DEBUG: Stored the security token in the session. {"key":"_security_main"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"cnf"},"url":"/en/category/11/parts","ip":"100.91.100.100","http_method":"GET","server":"parts.mydomain.com","referrer":null}
172.30.32.1 - - [23/Mar/2025:18:10:25 +0100] "GET /en/category/11/parts HTTP/1.1" 500 5973 "-" "Mozilla/5.0 (X11; Linux x86_64; rv:136.0) Gecko/20100101 Firefox/136.0"
- -  23/Mar/2025:17:10:25 +0000 "GET /index.php" 500
- -  23/Mar/2025:17:10:25 +0000 "POST /index.php" 204
172.30.32.1 - - [23/Mar/2025:18:10:25 +0100] "POST /csp/report HTTP/1.1" 204 859 "-" "Mozilla/5.0 (X11; Linux x86_64; rv:136.0) Gecko/20100101 Firefox/136.0"
- -  23/Mar/2025:17:12:38 +0000 "GET /index.php" 200
172.30.32.1 - - [23/Mar/2025:18:12:38 +0100] "GET /en/category/1/parts HTTP/1.1" 200 16341 "https://parts.mydomain.com/en/part/7/info" "Mozilla/5.0 (X11; Linux x86_64; rv:136.0) Gecko/20100101 Firefox/136.0"
- -  23/Mar/2025:17:12:41 +0000 "GET /index.php" 200
172.30.32.1 - - [23/Mar/2025:18:12:41 +0100] "GET /en/part/7/edit HTTP/1.1" 200 19761 "https://parts.mydomain.com/en/part/7/info" "Mozilla/5.0 (X11; Linux x86_64; rv:136.0) Gecko/20100101 Firefox/136.0"
- -  23/Mar/2025:17:14:22 +0000 "GET /index.php" 200
172.30.32.1 - - [23/Mar/2025:18:14:22 +0100] "GET /en/tools/server_infos HTTP/1.1" 200 9200 "https://parts.mydomain.com/en/part/7/edit" "Mozilla/5.0 (X11; Linux x86_64; rv:136.0) Gecko/20100101 Firefox/136.0"
```

cnf, 2025-03-23:
[image]

It seems to be image related. Why do we get error 500's when part-db can't load an external image?
And is there a way to run through and download all the ones that were not downloaded yet?

jbtronics, 2025-03-27:
This seemed to be some edge case, caused, when the internal path of the attachment, contained some invalid data somehow, that was completly unresolvable.

This should be fixed now however.

closed 2025-03-27 by jbtronics

cnf, 2025-03-28:
Nice, are you making a release for this? or is it just master until the next release?

Thanks!

cnf, 2025-03-29:
Can confirm issue is gone in master.

Can also confirm me losing the images was my doing, didn't map the volumes right, so when I restarted the container to change options, i would lose data.