Part-DB/Part-DB-server #901: Preview image not showing in parts list but showing in part detail view

SemvdH, 2025-03-30:
**Describe the bug**
A clear and concise description of what the bug is.
After uploading an image for a part, the preview image (from the cache) does not get shown. When opening the detail view for that part, the image gets shown perfectly fine.

I also noticed that the preview image gets prefixed with the IP of my server, instead of just using relative paths that are used in the detail view.
Preview image: 
```html
<img alt="Part image" src="//10.10.100.55:8991/media/cache/thumbnail_sm/media/part/win_20250329_22_59_31_projpg-67e875bc487e8.jpg" data-thumbnail="/media/cache/thumbnail_md/media/part/win_20250329_22_59_31_projpg-67e875bc487e8.jpg" class="hoverpic part-table-image" data-title="WIN_20250329_22_59_31_Pro.jpg (WIN_20250329_22_59_31_Pro.jpg)" data-controller="elements--hoverpic">
```
Detail image:
```html
<img class="d-block w-100 img-fluid img-thumbnail bg-light part-info-image" src="/media/part/win_20250329_22_59_31_projpg-67e875bc487e8.jpg" alt="">
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to parts list
2. See that the preview images are not shown
3. Click on a part
4. See that the image is shown

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect the preview images to be shown

**Screenshots**
Parts list:
[image]
Detail view:
[image]

**Server Side**
- Part-DB Version: [e.g. Part-DB 1.0 (for dev versions commit hash is useful] 1.16.1
- PHP Version: [e.g. PHP 7.4.3] 8.3.17
- Database Server [e.g. MySQL 5.7] ?? Couldn't find in the docs how I can get this info, symfony is 6.4.18. I'm using SQLite

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows 11 Home
 - Browser [e.g. chrome, safari] Vivaldi Browser, Microsoft Edge
 - Version [e.g. 22] Vivaldi 7.3.3635.4 (Stable channel) (64-bits), Edge 134.0.3124.93

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

I'm using docker, and have put my part-db instance behind nginx as reverse proxy, I have set the TRUSTED_PROXIES to the IP of my reverse proxy, and set the DEFAULT_URI to the URL to my part-db instance (with a slash at the end)

output of partdb:version: 
[OK] Part-DB version: 1.16.1

 [INFO] PHP version: 8.3.17

 [INFO] Symfony version: 6.4.18

 [INFO] OS: Linux b8a5fd4f03a1 6.8.12-2-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-2
        (2024-09-05T10:03Z) x86_64

 [INFO] PHP extension: Core, date, libxml, openssl, pcre, zlib, filter, hash,
        json, pcntl, random, Reflection, SPL, session, standard, sodium,
        mysqlnd, PDO, xml, bcmath, calendar, ctype, curl, dom, mbstring, FFI,
        fileinfo, ftp, gd, gettext, iconv, intl, exif, mysqli, pdo_mysql,
        pdo_pgsql, pdo_sqlite, pgsql, Phar, posix, readline, shmop, SimpleXML,
        sockets, sqlite3, sysvmsg, sysvsem, sysvshm, tokenizer, xmlreader,
        xmlwriter, xsl, zip, Zend OPcache

jbtronics, 2025-03-30:
Do your reverse proxy sets the correct headers? Especially X-Forwarded-Host?

SemvdH, 2025-03-30:
> Do your reverse proxy sets the correct headers? Especially X-Forwarded-Host?

No, it doesn't. What do I need to set as the X-Forwarded-Host? The docs only mention the X-Forwarded-Prefix if the part-db instance is behind a subpath

SemvdH, 2025-03-30:
I've fount [this post](https://stackoverflow.com/questions/74644096/modify-the-host-header-or-pass-x-forwarded-host-when-using-nginx-as-a-revers) and added the line `proxy_set_header X-Forwarded-Host $host;` to my nginx config file and now the images are show, Thanks!

closed 2025-03-30 by SemvdH