Part-DB/Part-DB-server #909: Upgrade from 1.15.1 to 1.17.0 leads to " Invalid or not existing database schema."

MG-5, 2025-04-10:
**Describe the bug**
Upgrade from 1.15.1 to 1.17.0 leads to " Invalid or not existing database schema."

[image]

**Server Side**
- Part-DB Version: 1.17.0 docker
- PHP Version: docker-php
- Database Server 11.4.5-MariaDB-log Alpine Linux (docker from linuxserver)

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu 24.04 Budgie
 - Browser: Firefox

**Additional context**
mysql -u "partdbuser" -p[password]

[image]

MG-5, 2025-04-10:
Did the database migration:migrate after docker-image change.
The database exists. The log files doesnt say any errors.

MG-5, 2025-04-13:
Digging in logs again, i saw an error line:
`[previous exception] [object] (PDOException(code: 42S22): SQLSTATE[42S22]: Column not found: 1054 Unknown column 't0.internal_path' in 'SELECT' at /var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Statement.php:55)
partdb         | [stacktrace]`

MG-5, 2025-04-13:
I create a new partdb-instance with MariaDB, create the database and grant the right privileges, i get a similar error while calling something from the left bar:
`[previous exception] [object] (PDOException(code: 42S22): SQLSTATE[42S22]: Column not found: 1054 Unknown column 'a2_.internal_path' in 'SELECT' at /var/www/html/vendor/doctrine/dbal/src/Driver/PDO/Statement.php:55)
partdb     | [stacktrace]
`

MG-5, 2025-04-13:
Revert docker image back to v.1.15.1 fix the issue for me, so there must be an migration issue for mariadb databases.

MG-5, 2025-04-13:
Migration from 1.15.1 to 1.17.0 no errors but also no activity to database:

[image]

jbtronics, 2025-04-13:
If you are running the migration command inside the container with sudo, you have to use the `-E` flag on sudo (like `sudo -E -u www bin/console`).

Otherwise it don't know about your database configuration for your container and it creates a new sqlite database (as you can see in the migration output).

MG-5, 2025-04-13:
Damn, make sense....

But sadly it doenst fix the problem.  Whether 
`sudo -E -u www-data bin/console doctrine:migrations:migrate` insider container or `sudo docker exec --user=www-data partdb php bin/console doctrine:migrations:migrate` outside container, this is what happens:

[image]

MG-5, 2025-04-14:
Hmm strange that this worked before without any problems. MariaDB must have become stricter.

Anyway, the problem is gone now!
I executed the command “ SET GLOBAL log_bin_trust_function_creators = 1;” in MariaDB as root.

closed 2025-04-14 by MG-5