Part-DB/Part-DB-server #910: Add custom fields to KiCad interface to support KiCost

Kampi, 2025-04-10:
**Describe the solution you'd like**

When using the KiCost plugin (i.e., with KiBot), you need the fields `manf`, `manf#`, and distributor-specific fields like `mouser#` to add the manufacturer, the part number, and the order number for the part. KiCost parses these fields to get the costs for specific components. The result looks like this.

[image]

I want to combine KiCost with the components from the KiCad interface, so either there must be a way to add custom fields to the interface, or the existing fields must be changed.

jbtronics, 2025-07-06:
Is there any way to directly pass a price (and a generic distributor and distributor ID to KiCost?), then you could use the information directly from Part-DB.

jbtronics, 2025-07-06:
Part-DB now adds the `mpn`, `mpn#` and a `supplier#` (like `digikey#`) fields in the API response, so that should now work with KiCost.