Part-DB/Part-DB-server #919: Error 500 when I click on a part

chr1st1an7, 2025-04-24:
**Describe the bug**
A clear and concise description of what the bug is.
Whenever I try to click on a part and open it, it gives a error 500. The logs say something about the digikey provider. I've been having issues with the database recently, now it seems to connect normally but now I'm having this problem. Logs are shown below. My Part-DB instance and the database are running in Docker.

**To Reproduce**
Steps to reproduce the behavior:
Try to open a part, and the error 500 is shown.

**Server Side**
- Part-DB Version: [e.g. Part-DB 1.0 (for dev versions commit hash is useful]  1.17.0
- PHP Version: [e.g. PHP 7.4.3] 8.3.19
- Database Server [e.g. MySQL 5.7]  MySQL 8.0.42

**Additional context**
Add any other context about the problem here.
Logs  from the docker container.
`partdb           | "} {"token":null,"url":"/en/part/330/info","ip":"172.18.0.2","http_method":"GET","server":"warehouse.chriggle.com","referrer":"https://warehouse.chriggle.com/en/category/4/parts"}
partdb           | [2025-04-24T18:50:30.820442+00:00] request.CRITICAL: Uncaught PHP Exception Twig\Error\RuntimeError: "An exception has been thrown during the rendering of a template ("Provider with key digikey already registered")." at _extended_infos.html.twig line 73 {"exception":"[object] (Twig\\Error\\RuntimeError(code: 0): An exception has been thrown during the rendering of a template (\"Provider with key digikey already registered\"). at /var/www/html/templates/parts/info/_extended_infos.html.twig:73)
partdb           | [stacktrace]
partdb           | #0 /var/www/html/var/cache/docker/twig/af/afab090124cf1d11f1cf19a050a992a9.php(464): Twig\\Template->yield()
partdb           | #1 /var/www/html/vendor/twig/twig/src/Template.php(432): __TwigTemplate_28e2e5c8bda01503431df505ccb0d91a->block_card_content()
partdb           | #2 /var/www/html/var/cache/docker/twig/a0/a0e74914a0b0d368e58dfb38ffc401a9.php(168): Twig\\Template->yieldBlock()
partdb           | #3 /var/www/html/vendor/twig/twig/src/Template.php(432): __TwigTemplate_8c5629aed6680842d1309abfc15854b3->block_card_body()
partdb           | #4 /var/www/html/var/cache/docker/twig/a0/a0e74914a0b0d368e58dfb38ffc401a9.php(80): Twig\\Template->yieldBlock()
partdb           | #5 /var/www/html/vendor/twig/twig/src/Template.php(432): __TwigTemplate_8c5629aed6680842d1309abfc15854b3->block_content()
partdb           | #6 /var/www/html/var/cache/docker/twig/0c/0ce5d3db9dd439df3f057c818e284615.php(297): Twig\\Template->yieldBlock()
partdb           | #7 /var/www/html/vendor/twig/twig/src/Template.php(432): __TwigTemplate_ba877508bb3bbc32ecbb12cf42786a0d->block_body()
partdb           | #8 /var/www/html/var/cache/docker/twig/0c/0ce5d3db9dd439df3f057c818e284615.php(138): Twig\\Template->yieldBlock()
partdb           | #9 /var/www/html/vendor/twig/twig/src/Template.php(388): __TwigTemplate_ba877508bb3bbc32ecbb12cf42786a0d->doDisplay()
partdb           | #10 /var/www/html/var/cache/docker/twig/a0/a0e74914a0b0d368e58dfb38ffc401a9.php(56): Twig\\Template->yield()
partdb           | #11 /var/www/html/vendor/twig/twig/src/Template.php(388): __TwigTemplate_8c5629aed6680842d1309abfc15854b3->doDisplay()
partdb           | #12 /var/www/html/var/cache/docker/twig/af/afab090124cf1d11f1cf19a050a992a9.php(50): Twig\\Template->yield()
partdb           | #13 /var/www/html/vendor/twig/twig/src/Template.php(388): __TwigTemplate_28e2e5c8bda01503431df505ccb0d91a->doDisplay()
partdb           | #14 /var/www/html/vendor/twig/twig/src/Template.php(344): Twig\\Template->yield()
partdb           | #15 /var/www/html/vendor/twig/twig/src/Template.php(359): Twig\\Template->display()
partdb           | #16 /var/www/html/vendor/twig/twig/src/TemplateWrapper.php(51): Twig\\Template->render()
partdb           | #17 /var/www/html/vendor/twig/twig/src/Environment.php(334): Twig\\TemplateWrapper->render()
partdb           | #18 /var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php(448): Twig\\Environment->render()
partdb           | #19 /var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php(453): Symfony\\Bundle\\FrameworkBundle\\Controller\\AbstractController->doRenderView()
partdb           | #20 /var/www/html/vendor/symfony/framework-bundle/Controller/AbstractController.php(253): Symfony\\Bundle\\FrameworkBundle\\Controller\\AbstractController->doRender()
partdb           | #21 /var/www/html/src/Controller/PartController.php(114): Symfony\\Bundle\\FrameworkBundle\\Controller\\AbstractController->render()
partdb           | #22 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(181): App\\Controller\\PartController->show()
partdb           | #23 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
partdb           | #24 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
partdb           | #25 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
partdb           | #26 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
partdb           | #27 /var/www/html/public/index.php(5): require_once('...')
partdb           | #28 {main}
partdb           | [previous exception] [object] (LogicException(code: 0): Provider with key digikey already registered at /var/www/html/src/Services/InfoProviderSystem/ProviderRegistry.php:75)
partdb           | [stacktrace]
partdb           | #0 /var/www/html/src/Services/InfoProviderSystem/ProviderRegistry.php(111): App\\Services\\InfoProviderSystem\\ProviderRegistry->initStructures()
partdb           | #1 /var/www/html/src/Twig/InfoProviderExtension.php(67): App\\Services\\InfoProviderSystem\\ProviderRegistry->getProviderByKey()
partdb           | #2 /var/www/html/var/cache/docker/twig/43/43bfe8e52dce915a9ec18815d8aec329.php(217): App\\Twig\\InfoProviderExtension->getInfoProviderName()
partdb           | #3 /var/www/html/vendor/twig/twig/src/Template.php(388): __TwigTemplate_c66e6e8607c518312e718dffee1d42bb->doDisplay()
partdb           | #4 /var/www/html/var/cache/docker/twig/af/afab090124cf1d11f1cf19a050a992a9.php(464): Twig\\Template->yield()
partdb           | #5 /var/www/html/vendor/twig/twig/src/Template.php(432): __TwigTemplate_28e2e5c8bda01503431df505ccb0d91a->block_card_content()
partdb           | #6 /var/www/html/var/cache/docker/twig/a0/a0e74914a0b0d368e58dfb38ffc401a9.php(168): Twig\\Template->yieldBlock()
partdb           | #7 /var/www/html/vendor/twig/twig/src/Template.php(432): __TwigTemplate_8c5629aed6680842d1309abfc15854b3->block_card_body()
partdb           | #8 /var/www/html/var/cache/docker/twig/a0/a0e74914a0b0d368e58dfb38ffc401a9.php(80): Twig\\Template->yieldBlock()
partdb           | #9 /var/www/html/vendor/twig/twig/src/Template.php(432): __TwigTemplate_8c5629aed6680842d1309abfc15854b3->block_content()
partdb           | #10 /var/www/html/var/cache/docker/twig/0c/0ce5d3db9dd439df3f057c818e284615.php(297): Twig\\Template->yieldBlock()
partdb           | #11 /var/www/html/vendor/twig/twig/src/Template.php(432): __TwigTemplate_ba877508bb3bbc32ecbb12cf42786a0d->block_body()
partdb           | #12 /var/www/html/var/cache/docker/twig/0c/0ce5d3db9dd439df3f057c818e284615.php(138): Twig\\Template->yieldBlock()
partdb           | #13 /var/www/html/vendor/twig/twig/src/Template.php(388): __TwigTemplate_ba877508bb3bbc32ecbb12cf42786a0d->doDisplay()
partdb           | #14 /var/www/html/var/cache/docker/twig/a0/a0e74914a0b0d368e58dfb38ffc401a9.php(56): Twig\\Template->yield()
partdb           | #15 /var/www/html/vendor/twig/twig/src/Template.php(388): __TwigTemplate_8c5629aed6680842d1309abfc15854b3->doDisplay()
partdb           | #16 /var/www/html/var/cache/docker/twig/af/afab090124cf1d11f1cf19a050a992a9.php(50): Twig\\Template->yield()
partdb           | #17 /var/www/html/vendor/twig/twig/src/Template.php(388): __TwigTemplate_28e2e5c8bda01503431df505ccb0d91a->doDisplay()
partdb           | #18 /var/www/html/vendor/twig/twig/src/Template.php(344): Twig\\Template->yield()
partdb           | #19 /var/www/html/vendor/twig/twig/src/Template.php(359): Twig\\Template->display()
[... 15 lines cut ...]
partdb           | [stacktrace]
[... cut ...]
partdb           | "} {"token":null,"url":"/en/part/330/info","ip":"172.18.0.2","http_method":"GET","server":"warehouse.chriggle.com","referrer":"https://warehouse.chriggle.com/en/category/4/parts"}
partdb           | 172.18.0.2 - - [24/Apr/2025:18:50:30 +0000] "GET /en/part/330/info HTTP/1.1" 500 5973 "https://warehouse.chriggle.com/en/category/4/parts" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/135.0.0.0 Safari/537.36"
partdb           | - -  24/Apr/2025:18:50:30 +0000 "GET /index.php" 500
`

jurviz, 2025-05-04:
Docker log from latest migration to version 1.16
[partdb-2025-05-04T12-27-11.log](https://github.com/user-attachments/files/20027500/partdb-2025-05-04T12-27-11.log)

jurviz, 2025-05-04:
I tried to do the migration after starting version 1.15 as described in the installation doc.
Output here:

[image]

After the migration the symptom changes and i get an 500 error immediately.

[image]

latest log is attached

[partdb-2025-05-04T12-56-01.log](https://github.com/user-attachments/files/20027632/partdb-2025-05-04T12-56-01.log)

kgalinec, 2025-05-04:
> Docker log from latest migration to version 1.16 [partdb-2025-05-04T12-27-11.log](https://github.com/user-attachments/files/20027500/partdb-2025-05-04T12-27-11.log)

Based on the logs you have a different error, mine said something about a missing Digikey provider even though I didn't use it... I fixed it by changing the code regarding the providers, and it seems to work fine now... Hopefully this will be fixed

jurviz, 2025-05-04:
I did a fallback to 1.15.2 with restored copy of sqlite db. I also tried to go migration path to 1.17 using the experimental image with another docker container, but can't get it to work either. The documentation is very sparse for this path.
I use part-db only for my hobby projects, so nothing critical.

jurviz, 2025-05-05:
The error is solved for me by installing version 1.17 in a new docker container with an empty sqlite db, stopping the container after db init and copying the database from the 1.15 volume into the new volume.
Automigration was set to true. After restart of the 1.17 container the application is working  now.

Basti-Fantasti, 2025-05-07:
This is also not working for me when try to search for parts using API calls like Mouser.
It works for the web scraping solutions

I'm on 1.17, too

closed 2026-05-25 by jbtronics