Part-DB/Part-DB-server #934: I am trying to import a file in the "BOM for projects" section but i don't use Kicad.

GabrieleFabi, 2025-05-12:
Hi, i wanted to import a BOM for projects but my company doesn't use Kicad, instead they use Altium. I am trying to find a way to import that file but up until now i wasn't able to. i didn't want to change the code, i was thinking of just finding a way to format the Alitum file to be accepted as a Kicad one, our file is a .xsls which i can always export as a CSV file. Anyone has an idea on how to format it or a way to go around it?

mkne, 2025-05-12:
I have uploaded a csv generated from the KiCad 9.0 demo project pic_programmer

It would be benificial for other users if you can share how you mapped the Altium BOM to the KiCad BOM. Maybe it would make a integration of a direct Altium BOM import possible.

[pic_programmer_en.csv](https://github.com/user-attachments/files/20164455/pic_programmer_en.csv)

Here is the link to the source file that does the import in PartDB, maybe this is more useful to understand what needs to be done:
https://github.com/Part-DB/Part-DB-server/blob/master/src/Services/ImportExportSystem/BOMImporter.php

turhanMust, 2025-05-14:
I had chatgpt write me a code to convert Altium BOM template to what partDB wants. I used Altium->Reports->Bill of Materials to create the BOM file. the required coloumns are: Designator,Footprint,Quantity,Mpn or equivalent(the code renames this to Designation.)

[bom_convert.txt](https://github.com/user-attachments/files/20208121/bom_convert.txt)

GabrieleFabi, 2025-05-14:
Hi. I hadn't yet mapped the Alitum BOM to the Kicad one but now after trying TurhanMust code i was able to convert it too and Part-DB accepted it. Thanks for the help.

closed 2025-05-14 by GabrieleFabi