Part-DB/Part-DB-server #935: 500 server error when trying to auth against Digikey

cnf, 2025-05-12:
**Describe the bug**
Similar to #581 , when I try to use the "connect oauth" button for digikey, I get a 500 error

**To Reproduce**
Steps to reproduce the behavior:
1) Enter the digikey key and secret in the docker container vars
2) go to /en/tools/info_providers/providers
3) click "connect oauth"

**Expected behavior**
authentication to succeed :P

**Screenshots**

[image]

**Server Side**
- Part-DB Version: 1.17.0 docker 
- PHP Version: 8.3.19 (SAPI: fpm-fcgi) docker image
- Database Server sqlite 3.40.1

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser firefox

**Additional context**
I have been poking at this on an off for a few weeks now,  It is the return call that is 500ing, after the auth  call to digikey succeeded.

It behaves exactly the same behind my reverse proxy on a domain, btw

jbtronics, 2025-05-12:
What does the Part-DB log tells?

This is most likely some configuration issue, that either one of the secrets is wrong, something is wrongly configured at digikey or something is wrong about the reverse proxy config. Digikey is quite picky about that.

cnf, 2025-05-12:
I'm not sure, my logs tend to get spammed with Symphony deprecation issues, and long tracebacks, it is kinda hard to find relevant entries in the noise. Any way to turn those down?

jbtronics, 2025-05-12:
Just search for "CRITICAL". Each log entry gets prefixed with its priority level, and only real errors get the CRITICAL priority.

cnf, 2025-05-12:
checked, the keys are the right ones.
Best I can tell from the browser network log, auth is succeeding, just not getting to part-db for some reason?
I have tried a few different browsers to make sure it's not some plugin, and even tried my ipad.
All give me the same issue.

henrikkoch, 2025-06-07:
I see the same problem.

EmilEmilchen, 2025-06-07:
Same for me:
```
[2025-06-07T21:43:05.096858+00:00] request.CRITICAL: Uncaught PHP Exception InvalidArgumentException: "Required option not passed: "access_token"" at AccessToken.php line 96 {"exception":"[object] (InvalidArgumentException(code: 0): Required option not passed: \"access_token\" at /var/www/html/vendor/league/oauth2-client/src/Token/AccessToken.php:96)
```

henrikkoch, 2025-06-09:
Solved ! 

Digikey has changed the length of both ID and secret. I created a new APP (and then got two new strings) and copied them in the ENV variables and replaced the old. Then I made a new OAuth connect:

https://<your URL>/en/tools/info_providers/providers

And everything now works again :-)

EmilEmilchen, 2025-06-09:
Weird, that didn't help for me, still getting
```
partdb  | [2025-06-09T22:07:20.489081+00:00] request.CRITICAL: Uncaught PHP Exception InvalidArgumentException: "Required option not passed: "access_token"" at AccessToken.php line 96 {"exception":"[object] (InvalidArgumentException(code: 0): Required option not passed: \"access_token\" at /var/www/html/vendor/league/oauth2-client/src/Token/AccessToken.php:96)
partdb  | [stacktrace]
partdb  | #0 /var/www/html/vendor/league/oauth2-client/src/Provider/AbstractProvider.php(858): League\\OAuth2\\Client\\Token\\AccessToken->__construct()
partdb  | #1 /var/www/html/vendor/league/oauth2-client/src/Provider/AbstractProvider.php(653): League\\OAuth2\\Client\\Provider\\AbstractProvider->createAccessToken()
partdb  | #2 /var/www/html/vendor/knpuniversity/oauth2-client-bundle/src/Client/OAuth2Client.php(106): League\\OAuth2\\Client\\Provider\\AbstractProvider->getAccessToken()
partdb  | #3 /var/www/html/src/Controller/OAuthClientController.php(60): KnpU\\OAuth2ClientBundle\\Client\\OAuth2Client->getAccessToken()
partdb  | #4 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(181): App\\Controller\\OAuthClientController->check()
partdb  | #5 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
partdb  | #6 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
partdb  | #7 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
partdb  | #8 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
partdb  | #9 /var/www/html/public/index.php(5): require_once('...')
partdb  | #10 {main}
```
despite having created a new DigiKey Production App. Callback URL in the digikey app config has to be empty though, right?

cnf, 2025-08-30:
I just re-created my key on digikey, and I am still getting the 500.
I am on 1.17.4

cnf, 2025-09-14:
2.x fixed this for me, btw

closed 2025-09-14 by jbtronics