Part-DB/Part-DB-server #937: Add inventory using scanner/barcode

ksjoberg, 2025-05-19:
Adding new inventory, such as the item in an delivered order is cumbersome, as the part need to be searched and/or added, edited, click "add stock" and then filling in the location and item count, followed by a click on Save.

The EIGP114 barcodes contain many of these datapoints. By scanning the bag, the part can be identified or added using the InfoProviderSystem. The added stock quantity is also in the EIGP114 data, so by picking the storage location the item could be quickly added.

afcruzb, 2025-05-20:
I would love this feature! I have a lot of LCSC parts, and if I could just scan the qr code to create the part, it would be awesome.

d-buchmann, 2025-08-01:
There is also (draft) PR #794 which adresses this kind of automation, but is currently limited to one certain barcode scanner type.
Image based scanning has been implemented already including EIGP114 decoding, therefore I'd propose to expand the "modes" of the scanner dialog to e.g. "Redirect mode" (the current default), "Info mode" and "New Entry mode".

RawSmokeTerribilus, 2025-11-16:
> I would love this feature! I have a lot of LCSC parts, and if I could just scan the qr code to create the part, it would be awesome.

I would love this feature too! actually what i am doing is to download the shopping cart (from lcsc, it has the option) when i order and edit the csv, then import. I actually use this method even for adding old stuff that i have lol. looking at the LCSC QR, yes... all the required info is there.