Part-DB/Part-DB-server #938: Inventory List - Critical Low

henrikkoch, 2025-05-21:
I would suggest a feature to list those stock items with lower than defined minimum (per component) in stock

In that way It maybe also could be exported as a comma separated list that could be directly added to a shópping cart at some of the already mentioned component dealers.

At the moment you are able to define a minimum quantity per component in PartDB - but as I see it you can't ask for a comment inventory list that show which items has reached this limit.

.

d-buchmann, 2025-05-21:
Try this:
`<your_partdb_instance>/parts?part_filter[lessThanDesired][value]=true&part_filter[submit]=`
I have added a link to this on my banner.md to access it quickly from the home page.
You can also filter parts by stocks and check the checkbox "Less than desired" manually.

d-buchmann, 2025-05-26:
No problem, I'm glad I could help.
Would you consider closing the issue or shall we leave it open until details like these have found their way into the docs?

henrikkoch, 2025-05-26:
I think we should keep it open.

It could be nice that this "feature" could be more integrated into a UI so
you don't have to make a filter. Could be nice in the future to have an
export file (comma separated) that could be used to import on Farnell,
Digikey etc. to put items in the basket.

/Henrik

d-buchmann, 2025-05-26:
Just a little brainstorming here:
What if there was a sub-item called "filters" in the "tools" sidebar menu, which contains some default filters?
That would be easily accessible for every user and it could be made customizable to fit the power users' needs, too.