Part-DB/Part-DB-server #954: Unable to do anything with projects on Fresh Install.

camcamnz, 2025-06-13:
**Describe the bug**
Unable to do anything with projects on Fresh Install. 
Seems like a UI bug.
Trying to import a BOM from KiCad as per the documentation, but I am unable to find anywhere to import, or create a new project manually. Clicking the "Projects" button on the UI navigates to the Parts URL.

**To Reproduce**
Steps to reproduce the behaviour:
1. Install new instance (following Docker-Compose instructions)
2. Try to create a project/import a BOM

**Expected behaviour**
Project page/menu to open

**Screenshots**

[image]

**Server Side**
- Part-DB Version: 1.17.1
- PHP Version: 8.3.21
- Database Server: sqlite 3.40.1

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Vivaldi
 - Version 7.4.3684.50 (Stable channel) (64-bit) 

**Additional context**
Probably not relevant, but additional steps I took between #1 and #2 were:
Changing - DEFAULT_TIMEZONE=Pacific/Auckland & - BASE_CURRENCY=NZD
Imported a csv of parts

camcamnz, 2025-06-13:
Also, just to add that I have also tried manually navigating to http://localhost:8080/en/projects but this gives a 404 error.
I have no idea if this is the correct URL though, if someone can tell me the correct one to try it would be greatly appreciated.

jbtronics, 2025-06-13:
To create a new project, you have to go to "Edit" -> "Projects" (https://demo.part-db.de/en/project/new). It seems that the root node of the projects menu is wrongly linked.

d-buchmann, 2025-06-27:
In fact, the root nodes of all data sources (except for tools) link to "all parts" instead of their own page. The behavior is even commented explicitly:

> src/Services/Trees/TreeViewGenerator.php:

```php
//We show the root node as a link to the list of all parts
$show_all_parts_url = $this->router->generate('parts_show_all');
```
I think it's not a bug but a usability issue, because it's not obvious to the new user how it works.
I issued PR #957 to adjust the links sensibly.