Part-DB/Part-DB-server #976: Initial install errors out with error 500 after login...

heinowalther, 2025-07-28:
Trying to getting this up and running on a setup with docker + swarm + portainer + traefik, using the stack feature...
Tried with both mysql and sqllite with the same result...
I use the local IP/port (to rule out traefik).
After running the initital database creation to get the admin password, I then try to login, but then get the error 500 page... 
Tried different browsers and incognito etc. with the same result..

Here is the output from the docker logs...

`- -  28/Jul/2025:11:36:47 +0000 "POST /index.php" 302
172.20.0.4 - - [28/Jul/2025:11:36:47 +0000] "POST /en/login HTTP/1.1" 302 969 "http://10.0.2.91:8881/en/login?_target_path=%2F" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/18.5 Safari/605.1.15"
- -  28/Jul/2025:11:36:48 +0000 "GET /index.php" 302
172.20.0.4 - - [28/Jul/2025:11:36:48 +0000] "GET / HTTP/1.1" 302 852 "http://10.0.2.91:8881/en/login?_target_path=%2F" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/18.5 Safari/605.1.15"
[2025-07-28T11:36:48.630235+00:00] request.INFO: Matched route "user_settings". {"route":"user_settings","route_parameters":{"_route":"user_settings","_controller":"App\\Controller\\UserSettingsController::userSettings","_locale":"en"},"request_uri":"http://10.0.2.91:8881/en/user/settings","method":"GET"} {"token":null,"url":"/en/user/settings","ip":"172.20.0.4","http_method":"GET","server":"10.0.2.91","referrer":"http://10.0.2.91:8881/en/login?_target_path=%2F"}
[2025-07-28T11:36:48.630840+00:00] deprecation.INFO: User Deprecated: Since api-platform/symfony 4.1: The "api_platform.validator.query_parameter_validator" service is deprecated use "\ApiPlatform\Metadata\Parameter::$constraints" instead. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since api-platform/symfony 4.1: The \"api_platform.validator.query_parameter_validator\" service is deprecated use \"\\ApiPlatform\\Metadata\\Parameter::$constraints\" instead. at /var/www/html/var/cache/docker/Container7c1LdZT/getApiPlatform_Validator_QueryParameterValidatorService.php:23)
[stacktrace]
#0 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(201): Container7c1LdZT\\getApiPlatform_Validator_QueryParameterValidatorService::do()
#1 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(1251): Container7c1LdZT\\App_KernelDockerContainer->load()
#2 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(345): Container7c1LdZT\\App_KernelDockerContainer::getApiPlatform_Listener_View_ValidateQueryParametersService()
#3 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(257): Container7c1LdZT\\App_KernelDockerContainer::Container7c1LdZT\\{closure}()
#4 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
#5 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
#6 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(157): Symfony\\Component\\EventDispatcher\\EventDispatcher->dispatch()
#7 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
#8 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#9 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#10 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#11 /var/www/html/public/index.php(5): require_once()
#12 {main}
"} {"token":null,"url":"/en/user/settings","ip":"172.20.0.4","http_method":"GET","server":"10.0.2.91","referrer":"http://10.0.2.91:8881/en/login?_target_path=%2F"}
[2025-07-28T11:36:48.631114+00:00] deprecation.INFO: User Deprecated: Since api-platform/core 3.4: The class "ApiPlatform\ParameterValidator\ParameterValidator" is deprecated, use "\ApiPlatform\Metadata\Parameter::$constraints" instead. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since api-platform/core 3.4: The class \"ApiPlatform\\ParameterValidator\\ParameterValidator\" is deprecated, use \"\\ApiPlatform\\Metadata\\Parameter::$constraints\" instead. at /var/www/html/vendor/api-platform/core/src/ParameterValidator/ParameterValidator.php:41)
[stacktrace]
#0 /var/www/html/var/cache/docker/Container7c1LdZT/getApiPlatform_Validator_QueryParameterValidatorService.php(31): ApiPlatform\\ParameterValidator\\ParameterValidator->__construct()
#1 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(201): Container7c1LdZT\\getApiPlatform_Validator_QueryParameterValidatorService::do()
#2 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(1251): Container7c1LdZT\\App_KernelDockerContainer->load()
#3 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(345): Container7c1LdZT\\App_KernelDockerContainer::getApiPlatform_Listener_View_ValidateQueryParametersService()
#4 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(257): Container7c1LdZT\\App_KernelDockerContainer::Container7c1LdZT\\{closure}()
#5 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
#6 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
#7 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(157): Symfony\\Component\\EventDispatcher\\EventDispatcher->dispatch()
#8 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
#9 /var/www/html/vendor/symfony/http-kernel/Kernel.php(197): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#10 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#11 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#12 /var/www/html/public/index.php(5): require_once()
#13 {main}
"} {"token":null,"url":"/en/user/settings","ip":"172.20.0.4","http_method":"GET","server":"10.0.2.91","referrer":"http://10.0.2.91:8881/en/login?_target_path=%2F"}
[2025-07-28T11:36:48.631250+00:00] deprecation.INFO: User Deprecated: Since api-platform/core 3.4: The class "ApiPlatform\ParameterValidator\Validator\ArrayItems" is deprecated, use "\ApiPlatform\Metadata\Parameter::$constraints" instead. {"exception":"[object] (ErrorException(code: 0): User Deprecated: Since api-platform/core 3.4: The class \"ApiPlatform\\ParameterValidator\\Validator\\ArrayItems\" is deprecated, use \"\\ApiPlatform\\Metadata\\Parameter::$constraints\" instead. at /var/www/html/vendor/api-platform/core/src/ParameterValidator/Validator/ArrayItems.php:25)
[stacktrace]
#0 /var/www/html/vendor/api-platform/core/src/ParameterValidator/ParameterValidator.php(46): ApiPlatform\\ParameterValidator\\Validator\\ArrayItems->__construct()
#1 /var/www/html/var/cache/docker/Container7c1LdZT/getApiPlatform_Validator_QueryParameterValidatorService.php(31): ApiPlatform\\ParameterValidator\\ParameterValidator->__construct()
#2 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(201): Container7c1LdZT\\getApiPlatform_Validator_QueryParameterValidatorService::do()
#3 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(1251): Container7c1LdZT\\App_KernelDockerContainer->load()
#4 /var/www/html/var/cache/docker/Container7c1LdZT/App_KernelDockerContainer.php(345): Container7c1LdZT\\App_KernelDockerContainer::getApiPlatform_Listener_View_ValidateQueryParametersService()
#5 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(257): Container7c1LdZT\\App_KernelDockerContainer::Container7c1LdZT\\{closure}()
#6 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(220): Symfony\\Component\\EventDispatcher\\EventDispatcher::Symfony\\Component\\EventDispatcher\\{closure}()
#7 /var/www/html/vendor/symfony/event-dispatcher/EventDispatcher.php(56): Symfony\\Component\\EventDispatcher\\EventDispatcher->callListeners()
[... cut ...] tings","ip":"172.20.0.4","http_method":"GET","server":"10.0.2.91","referrer":"http://10.0.2.91:8881/en/login?_target_path=%2F"}
172.20.0.4 - - [28/Jul/2025:11:36:48 +0000] "GET /en/user/settings HTTP/1.1" 500 4367 "http://10.0.2.91:8881/en/login?_target_path=%2F" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/18.5 Safari/605.1.15"
- -  28/Jul/2025:11:36:48 +0000 "GET /index.php" 500
172.20.0.4 - - [28/Jul/2025:11:36:58 +0000] "-" 408 0 "-" "-"
`

heinowalther, 2025-07-28:
Update...  it seems that if I get it runnint with "EUR", and the try changing it to "DKK" just once...  it breaks it self... and even if I put it back to "EUR" again i keep getting the 500 error...  is this a "once in a lifetime" option? ;-)

jbtronics, 2025-08-12:
You have to set the correct ISO codes for the currency otherwise it will fail.

In general you shouldnt change the `BASE_CURRENCY` in a database filled with data, as the prices will be wrong afterwards most likely. But it should not throw 500 errors from the changing itself.

What error messages do you get exactly?