Part-DB/Part-DB-server #977: Export from MySQL to MariaDB - Help!

henrikkoch, 2025-07-29:
Can anyone please help with a recepi of how to move all data from the MySQL db to a MariaDB

I have tried to export and import it to a MariaDB. 

After this I have changed the PartDB DB path to the new with the correct logins. But this does not work. 

It could be nice if we could get some kind of scripts that could to this export/import correct.

Have anybody succeeded with this?

virtadpt, 2025-07-30:
Do you mean that you ran a `mysqldump` and then sourced the .sql file in a `mariadb` session?  Or something else?

henrikkoch, 2025-07-30:
I used an online tool which could import the MySQL file and convert it to MariaDB. Like this one:  https://www.rebasedata.com/convert-sql-to-mariadb-online

jbtronics, 2025-07-30:
What errors does MariaDB throw when trying to import the SQL dump?
MariaDB and MySQL should be so similar, that i would not expect any issues. Or if there are ones these should be fixable by relaxing some SQL_MODE settings a bit.

virtadpt, 2025-07-31:
@jbtronics That's exactly what I was thinking.  MariaDB is a fork of MySQL.  And I've imported MySQL database dumps into MariaDB servers a few times now without incident.  A significant number of webapps written in PHP interface with MariaDB without incident as well (or at least, all the ones I've thrown at the last couple of major releases).