Part-DB/Part-DB-server #1006: Server unable to start after v2.0.x update

n-rosati, 2025-08-31:
**Describe the bug**
After upgradign from 1.17.4 -> 2.0.0, the server no longer starts and only a "504 Service Unavailable" page is returned. In the console logs on a page reload it mentions not being able to connect to the php8.4-fpm.sock socket. A little investigation returned that the php8.4-fpm service didn't start (and after trying to manually start it, does not start). I tried to look around for PHP logs but I couldn't find any, though I am totally unfamiliar with PHP and how it behaves.

Logs:

```
Waiting for database to be ready...
The database is now ready and reachable
sudo: unable to send audit message: Operation not permitted

 [OK] Up-to-date! No migrations to execute.

Database is up to date, no migrations necessary.
AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 172.16.0.100. Set the 'ServerName' directive globally to suppress this message
AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 172.16.0.100. Set the 'ServerName' directive globally to suppress this message
[Sun Aug 31 02:35:14.105501 2025] [mpm_event:notice] [pid 1:tid 1] AH00489: Apache/2.4.62 (Debian) configured -- resuming normal operations
[Sun Aug 31 02:35:14.105709 2025] [core:notice] [pid 1:tid 1] AH00094: Command line: 'apache2 -D FOREGROUND'
[Sun Aug 31 02:35:38.151090 2025] [proxy:error] [pid 49:tid 53] (2)No such file or directory: AH02454: FCGI: attempt to connect to Unix domain socket /run/php/php8.4-fpm.sock (localhost:8000) failed
[Sun Aug 31 02:35:38.151153 2025] [proxy_fcgi:error] [pid 49:tid 53] [client 172.16.0.160:39082] AH01079: failed to make connection to backend: localhost
172.16.0.160 - - [31/Aug/2025:02:35:38 +0000] "GET / HTTP/1.1" 503 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:140.0) Gecko/20100101 Firefox/140.0"
172.16.0.160 - - [31/Aug/2025:02:35:38 +0000] "GET /favicon.ico HTTP/1.1" 200 15333 "https://partdb.[DOMAIN].xyz/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:140.0) Gecko/20100101 Firefox/140.0"
```

**To Reproduce**
Unsure if reproducible but my steps taken were:
- update image tag to v2.0.0 (later v2.0.1)
- wait for redeployment with updated image
- once pod is running, navigate to the Part-DB instance

**Expected behavior**
The server to load when navigating to it

**Server Side**
- Part-DB Version: 2.0.1
- PHP Version: [e.g. PHP 7.4.3]
- Database Server Postgres 16

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Waterfox
 - Version 6.6.1

**Additional context**
Running in a Kubernetes cluster (config [here](https://github.com/n-rosati/homelab-infra/blob/main/clusters/main/kubernetes/apps/part-db/app/helm-release.yaml) )

drpetersen, 2025-08-31:
I get the same error, except it says 

```
ERROR: [/etc/php/8.4/fpm/pool.d/99-env.conf:14] You must provide a key for field 'env'
```

instead of 

```
ERROR: [/etc/php/8.4/fpm/pool.d/99-env.conf:9] empty value
```

And indeed, `99-env.conf` contains an empty value on line 14:

```
; auto-generated env config
[www]
clear_env = no
env[DEFAULT_LANG] = "de"
env[DATABASE_URL] = "mysql://partdb:SECRETPASSWORD@mysql.MYDOMAIN.de:3306/partdb"
env[SAML_SP_PRIVATE_KEY] = "MIIJKQIBA…very long secret…wesMyCr"
env[APACHE_CONFDIR] = "/etc/apache2"
env[INSTANCE_NAME] = "Part-DB"
env[SAML_SP_ENTITY_ID] = "https://partdb.MYDOMAIN.de/sp"
env[PROVIDER_TME_SECRET] = "MYSECRET"
env[HOME] = "/root"
env[SAML_BEHIND_PROXY] = "1"
env[BANNER] = "Welcome to Part-DB."
env[] = ""
env[SAML_SP_X509_CERT] = "MIIE4…very long secret …XFGiY"
(… snip …)
```

As it says, the file was auto-generated, no idea where that empty value originates from.

drpetersen, 2025-08-31:
So I tried to fix that, entering the container as root,

```
docker exec -ti partdb bash
```

deleted the offending line, and restarted php:

```
grep -v "env\[\]" /etc/php/8.4/fpm/pool.d/99-env.conf > /etc/php/8.4/fpm/pool.d/99-env.conf.new
diff /etc/php/8.4/fpm/pool.d/99-env.conf /etc/php/8.4/fpm/pool.d/99-env.conf.new
# just to be sure …
mv /etc/php/8.4/fpm/pool.d/99-env.conf.new /etc/php/8.4/fpm/pool.d/99-env.conf
/usr/sbin/php-fpm8.4 --nodaemonize --fpm-config /etc/php/8.4/fpm/php-fpm.conf
```

Now php starts, but there is another error:

```
[31-Aug-2025 09:48:38] NOTICE: fpm is running, pid 117
[31-Aug-2025 09:48:38] NOTICE: ready to handle connections
[31-Aug-2025 09:48:38] NOTICE: systemd monitor interval set to 10000ms
[2025-08-31T09:48:41.613932+00:00] deprecation.INFO: User Deprecated: Class "Doctrine\ORM\Proxy\Autoloader" is deprecated. Use native lazy objects instead. (Autoloader.php:74 called by DoctrineBundle.php:130, https://github.com/doctrine/orm/pull/12005, package doctrine/orm) {"exception":"[object] (ErrorException(code: 0): User Deprecated: Class \"Doctrine\\ORM\\Proxy\\Autoloader\" is deprecated. Use native lazy objects instead. (Autoloader.php:74 called by DoctrineBundle.php:130, https://github.com/doctrine/orm/pull/12005, package doctrine/orm) at /var/www/html/vendor/doctrine/deprecations/src/Deprecation.php:208)
[stacktrace]
#0 /var/www/html/vendor/doctrine/deprecations/src/Deprecation.php(108): Doctrine\\Deprecations\\Deprecation::delegateTriggerToBackend()
#1 /var/www/html/vendor/doctrine/orm/src/Proxy/Autoloader.php(74): Doctrine\\Deprecations\\Deprecation::trigger()
#2 /var/www/html/vendor/doctrine/doctrine-bundle/src/DoctrineBundle.php(130): Doctrine\\ORM\\Proxy\\Autoloader::register()
#3 /var/www/html/vendor/symfony/http-kernel/Kernel.php(125): Doctrine\\Bundle\\DoctrineBundle\\DoctrineBundle->boot()
#4 /var/www/html/vendor/symfony/http-kernel/Kernel.php(177): Symfony\\Component\\HttpKernel\\Kernel->boot()
#5 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#6 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#7 /var/www/html/public/index.php(5): require_once()
#8 {main}
"} {"token":null}
[2025-08-31T09:48:41.640511+00:00] request.INFO: Matched route "redirector". {"route":"redirector","route_parameters":{"_route":"redirector","_controller":"App\\Controller\\RedirectController::addLocalePart","url":""},"request_uri":"https://partdb.dataverse.de/","method":"GET"} {"token":null,"url":"/","ip":"87.79.248.211","http_method":"GET","server":"partdb.dataverse.de","referrer":null}
[2025-08-31T09:48:41.735142+00:00] request.CRITICAL: Uncaught PHP Exception Symfony\Component\DependencyInjection\Exception\RuntimeException: "Invalid JSON in env var "SAML_ROLE_MAPPING": Syntax error" at EnvVarProcessor.php line 294 {"exception":"[object] (Symfony\\Component\\DependencyInjection\\Exception\\RuntimeException(code: 0): Invalid JSON in env var \"SAML_ROLE_MAPPING\": Syntax error at /var/www/html/vendor/symfony/dependency-injection/EnvVarProcessor.php:294)
```

And indeed, that `SAML_ROLE_MAPPING` variable contains quotes that likely need to be escaped:

```
grep "SAML_ROLE_MAPPING" /etc/php/8.4/fpm/pool.d/99-env.conf
# results in:
env[SAML_ROLE_MAPPING] = "{"admin": 1, "partdb-readonly": 2, "*": 3}"
```

So I guess that requires some special treatment as well. Thank you so much for your time and effort!

MayNiklas, 2025-08-31:
I'm having the same issue.

drpetersen, 2025-08-31:
I can confirm that it generally works now after I changed the above line to 

```
env[SAML_ROLE_MAPPING] = "{\"admin\": 1, \"partdb-readonly\": 2, \"*\": 3}"
```

Obviously, any other environment variables containing quotes need to be escaped likewise. For me, there was no other such value.

If somebody else wants to try this, do not restart the docker after changing `/etc/php/8.4/fpm/pool.d/99-env.conf`, that will regenerate the file and overwrite your changes. Rather start php again with 

```/usr/sbin/php-fpm8.4 --nodaemonize --fpm-config /etc/php/8.4/fpm/php-fpm.conf```

Leave out the `--nodaemonize` if it runs ok. Oh, and you will probably want to `apt update && apt install nano` or some other editor for editing the config file.

Now the next error in my case is an Authentication failure due to the SAML key not being retrieved for some reason. As I can login locally with the admin account, I think this is a different problem. The environment variables `SAML_SP_PRIVATE_KEY`, `SAML_SP_X509_CERT` and `SAML_IDP_X509_CERT` all look fine in `/etc/php/8.4/fpm/pool.d/99-env.conf`. So I would rather open another issue later, but only after this issue here is fixed. (Or should I rather post the error messages right here?)

n-rosati, 2025-08-31:
After fixing up the 99-env.conf file and starting php-fpm it seems to be working again. 
I was using the default banner message which contained `\nabla` and caused issues as the `\n` portion was recognized as a new line. Any of the terms starting with a `\` just had to be escaped.

However I noticed a further issue where `/etc/php/8.4/fpm/pool.d/99-env.conf` does not persist or get regenerated after a container restart. It will get generated on v2.0.0 but not v2.0.1.

jbtronics, 2025-08-31:
For the docker image 2.0.2 a different way to pass the env to part DB is used and the 99-env is not used and not required anymore.