Part-DB/Part-DB-server #1007: Invalid CSRF token when login after the update

Kampi, 2025-08-31:
**Describe the bug**

[image]

**Server Side**
- Part-DB Version: latest via Docker

**Additional context**

```
2025/08/31 12:23:25,stdout,"172.19.0.1 - - [31/Aug/2025:10:23:20 +0000] \"GET /de/ HTTP/1.1\" 200 7987 \"-\" \"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/139.0.0.0 Safari/537.36\"
2025/08/31 12:23:25,stdout,- -  31/Aug/2025:10:23:20 +0000 "GET /index.php" 200
2025/08/31 12:23:20,stdout,"172.19.0.1 - - [31/Aug/2025:10:23:10 +0000] \"GET / HTTP/1.1\" 302 826 \"-\" \"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/139.0.0.0 Safari/537.36\"
2025/08/31 12:23:20,stdout,- -  31/Aug/2025:10:23:11 +0000 "GET /index.php" 302
2025/08/31 12:23:02,stderr,[Sun Aug 31 10:23:02.733817 2025] [core:notice] [pid 1:tid 1] AH00094: Command line: 'apache2 -D FOREGROUND'
2025/08/31 12:23:02,stderr,[Sun Aug 31 10:23:02.733201 2025] [mpm_event:notice] [pid 1:tid 1] AH00489: Apache/2.4.62 (Debian) configured -- resuming normal operations
2025/08/31 12:23:02,stderr,"AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 172.19.0.2. Set the 'ServerName' directive globally to suppress this message
2025/08/31 12:23:02,stderr,"AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 172.19.0.2. Set the 'ServerName' directive globally to suppress this message
2025/08/31 12:23:02,stdout,[31-Aug-2025 10:23:02] NOTICE: systemd monitor interval set to 10000ms
2025/08/31 12:23:02,stdout,[31-Aug-2025 10:23:02] NOTICE: ready to handle connections
2025/08/31 12:23:02,stdout,"[31-Aug-2025 10:23:02] NOTICE: fpm is running, pid 24
```

Kampi, 2025-08-31:
> Try to restart the docker container, clear cache inside the docker container and ensure that the browser do not cache any old scripts (like try to login in an incognito tab or something).
> 
> Also check if you have an adblocker active that might interfere with some scripts on Part-DB.
> 
> Check your browser devloper console for any error messages.

I´ve purged the Docker container, built it new, tried an incognito session in Chrome and pause my Adblocker. I get this error message

[image]

cansi22, 2025-08-31:
Same error
Update to 2.0.2 in docker.

Error in computer and mobile with 4G network

Kampi, 2025-08-31:
> I assume you are running through a reverse proxy. Setting the **TRUSTED_PROXIES** variable helped me

I´ve run it behind a proxy before the update without issues and added the variable now. But no change.

leroyloren, 2025-08-31:
`TRUSTED_PROXIES=127.0.0.0/8,::1,10.0.0.0/8,172.16.0.0/12,192.168.0.0/16`

Example from doc

cansi22, 2025-08-31:
> `TRUSTED_PROXIES=127.0.0.0/8,::1,10.0.0.0/8,172.16.0.0/12,192.168.0.0/16`
> 
> Example from doc

With this ENV settings it's works for me

Kampi, 2025-08-31:
> `TRUSTED_PROXIES=127.0.0.0/8,::1,10.0.0.0/8,172.16.0.0/12,192.168.0.0/16`
> 
> Example from doc

Same

closed 2025-08-31 by Kampi