Part-DB/Part-DB-server #1010: Migrating: 1.17.4 -> 2.0.2 Uncaught Error: Class "Jbtronics\SettingsBundle\JbtronicsSettingsBundle" not found

Tropaion, 2025-09-01:
**Describe the bug**
When upgrading 1.17.4 -> 2.0.2 and running the migration script, I get this error:
```
root@PartDB:/var/www/partdb# sudo -u www-data php bin/console doctrine:migrations:migrate
Symfony\Component\ErrorHandler\Error\ClassNotFoundError^ {#87
  #message: """
    Attempted to load class "JbtronicsSettingsBundle" from namespace "Jbtronics\SettingsBundle".\n
    Did you forget a "use" statement for another namespace?
    """
  #code: 0
  #file: "/var/www/partdb/vendor/symfony/framework-bundle/Kernel/MicroKernelTrait.php"
  #line: 136
  trace: {
    /var/www/partdb/vendor/symfony/framework-bundle/Kernel/MicroKernelTrait.php:136 { …}
    /var/www/partdb/vendor/symfony/http-kernel/Kernel.php:346 { …}
    /var/www/partdb/vendor/symfony/http-kernel/Kernel.php:771 { …}
    /var/www/partdb/vendor/symfony/http-kernel/Kernel.php:126 { …}
    /var/www/partdb/vendor/symfony/framework-bundle/Console/Application.php:190 { …}
    /var/www/partdb/vendor/symfony/framework-bundle/Console/Application.php:72 { …}
    /var/www/partdb/vendor/symfony/console/Application.php:175 { …}
    /var/www/partdb/vendor/symfony/runtime/Runner/Symfony/ConsoleApplicationRunner.php:49 { …}
    /var/www/partdb/vendor/autoload_runtime.php:29 { …}
    /var/www/partdb/bin/console:22 {
      › 
      › require_once dirname(__DIR__).'/vendor/autoload_runtime.php';
      › 
      arguments: {
        "/var/www/partdb/vendor/autoload_runtime.php"
      }
    }
  }
}
2025-09-01T11:12:42+00:00 [critical] Uncaught Error: Class "Jbtronics\SettingsBundle\JbtronicsSettingsBundle" not found```

**To Reproduce**
Steps to reproduce the behavior:
I followed the migration guide until:
```sudo -u www-data php bin/console doctrine:migrations:migrate```

**Server Side**
- Part-DB Version: 1.17.4 -> 2.0.2
- PHP Version: 8.4.11
- Database Server: 10.11.14-MariaDB-deb11

jbtronics, 2025-09-01:
Have you updated the dependencies before with composer install?

Tropaion, 2025-09-01:
I think I did, but maybe I missed it.
I tried it and got this:
```
root@PartDB:/var/www/partdb# sudo -u www-data composer install --no-dev -o
Deprecation Notice: Symfony\Component\Console\Command\Command::__construct(): Implicitly marking parameter $name as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Command/Command.php:99
Deprecation Notice: Symfony\Component\Console\Command\Command::setApplication(): Implicitly marking parameter $application as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Command/Command.php:135
Deprecation Notice: Symfony\Component\Console\Command\Command::addArgument(): Implicitly marking parameter $mode as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Command/Command.php:436
Deprecation Notice: Symfony\Component\Console\Command\Command::addOption(): Implicitly marking parameter $mode as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Command/Command.php:457
Deprecation Notice: Symfony\Component\Console\Input\InputArgument::__construct(): Implicitly marking parameter $mode as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Input/InputArgument.php:41
Deprecation Notice: Symfony\Component\Console\Input\InputOption::__construct(): Implicitly marking parameter $mode as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Input/InputOption.php:62
Deprecation Notice: Symfony\Component\Console\Helper\ProcessHelper::run(): Implicitly marking parameter $error as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Helper/ProcessHelper.php:35
Deprecation Notice: Symfony\Component\Console\Helper\ProcessHelper::run(): Implicitly marking parameter $callback as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Helper/ProcessHelper.php:35
Deprecation Notice: Symfony\Component\Console\Helper\ProcessHelper::mustRun(): Implicitly marking parameter $error as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Helper/ProcessHelper.php:101
Deprecation Notice: Symfony\Component\Console\Helper\ProcessHelper::mustRun(): Implicitly marking parameter $callback as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Helper/ProcessHelper.php:101
Deprecation Notice: Symfony\Component\Console\Helper\ProcessHelper::wrapCallback(): Implicitly marking parameter $callback as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/symfony/console/Helper/ProcessHelper.php:115
Deprecation Notice: Constant E_STRICT is deprecated in phar:///usr/local/bin/composer/src/Composer/Util/Silencer.php:36
Deprecation Notice: Constant E_STRICT is deprecated in phar:///usr/local/bin/composer/src/Composer/Util/Silencer.php:36
Deprecation Notice: Composer\Pcre\Preg::replace(): Implicitly marking parameter $count as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/composer/pcre/src/Preg.php:111
Deprecation Notice: Composer\Pcre\Preg::replaceCallback(): Implicitly marking parameter $count as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/composer/pcre/src/Preg.php:135
Deprecation Notice: Composer\Pcre\Preg::replaceCallbackArray(): Implicitly marking parameter $count as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/composer/pcre/src/Preg.php:163
Deprecation Notice: JsonSchema\Constraints\BaseConstraint::__construct(): Implicitly marking parameter $factory as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/justinrainbow/json-schema/src/JsonSchema/Constraints/BaseConstraint.php:41
Deprecation Notice: JsonSchema\Constraints\BaseConstraint::addError(): Implicitly marking parameter $path as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/justinrainbow/json-schema/src/JsonSchema/Constraints/BaseConstraint.php:46
Deprecation Notice: JsonSchema\Constraints\BaseConstraint::addError(): Implicitly marking parameter $more as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/justinrainbow/json-schema/src/JsonSchema/Constraints/BaseConstraint.php:46
Deprecation Notice: JsonSchema\Constraints\Factory::__construct(): Implicitly marking parameter $schemaStorage as nullable is deprecated, the explicit nullable type must be used instead in phar:///usr/local/bin/composer/vendor/justinrainbow/json-schema/src/JsonSchema/Constraints/Factory.php:76
[... cut ...]
Run composer recipes at any time to see the status of your Symfony recipes.

Executing script cache:clear [KO]
 [KO]
Script cache:clear returned with error code 1
!!  
!!  In App_KernelProdContainer.php line 2839:
!!                                                                                 
!!    You have requested a non-existent parameter "jbtronics.settings.proxy_dir".  
!!                                                                                 
!!  
!!  
Script @auto-scripts was called via post-install-cmd
```

Tropaion, 2025-09-01:
Thanks, composer is updated an the install process didn't throw any errors.
But when I open PartDB in the browser now I get this:
[image]
I have php 8.4 installed:
```
root@PartDB:~# php -v
PHP 8.4.11 (cli) (built: Aug  3 2025 08:46:38) (NTS)
Copyright (c) The PHP Group
Built by Debian
Zend Engine v4.4.11, Copyright (c) Zend Technologies
    with Zend OPcache v8.4.11, Copyright (c), by Zend Technologies
```
And composer also says it's using it:
```
root@PartDB:/var/www/partdb# composer -vvv about
Running 2.8.11 (2025-08-21 11:29:39) with PHP 8.4.11 on Linux / 6.8.12-13-pve
Do not run Composer as root/super user! See https://getcomposer.org/root for details
Continue as root/super user [yes]? yes
Reading ./composer.json (/var/www/partdb/composer.json)
Loading config file ./composer.json (/var/www/partdb/composer.json)
Checked CA file /etc/pki/ca-trust/extracted/pem/tls-ca-bundle.pem does not exist or it is not a file.
Checked directory /etc/pki/ca-trust/extracted/pem/tls-ca-bundle.pem does not exist or it is not a directory.
Checked CA file /etc/pki/tls/certs/ca-bundle.crt does not exist or it is not a file.
Checked directory /etc/pki/tls/certs/ca-bundle.crt does not exist or it is not a directory.
Checked CA file /etc/ssl/certs/ca-certificates.crt: valid
Executing command (/var/www/partdb): 'git' 'branch' '-a' '--no-color' '--no-abbrev' '-v'
Executing async command (/var/www/partdb): 'git' 'rev-list' 'master..a6be786d5dd12dcb58d8c5e65e0804da454142ad'
Executing async command (/var/www/partdb): 'git' 'rev-list' 'remotes/origin/master..a6be786d5dd12dcb58d8c5e65e0804da454142ad'
Executing command (/var/www/partdb): 'git' 'describe' '--exact-match' '--tags'
Failed to initialize global composer: Composer could not find the config file: /root/.config/composer/composer.json

Reading /var/www/partdb/vendor/composer/installed.json
Loading plugin Http\Discovery\Composer\Plugin (from php-http/discovery)
Loading plugin Symfony\Component\Runtime\Internal\ComposerPlugin (from symfony/runtime)
Loading plugin PackageVersions\Installer (from composer/package-versions-deprecated)
Loading plugin Symfony\Flex\Flex (from symfony/flex)
Composer - Dependency Manager for PHP - version 2.8.11
Composer is a dependency manager tracking local dependencies of your projects and libraries.
See https://getcomposer.org/ for more information.
```
But I can't make sense of the other errors here.

jbtronics, 2025-09-01:
Are your sure that your webserver is configured to use Part-DB 8.4 and not some other installed version?

Tropaion, 2025-09-01:
Do you mean php8.4 or PartDB 2.0.2?
I don't know much about composer, but in the log output it clearly says PHP 8.4.11 as posted above.
PartDB should also be correct:
```
root@PartDB:/var/www/partdb# git branch
* (HEAD detached at v2.0.2)
  master
root@PartDB:/var/www/partdb# ls
CODE_OF_CONDUCT.md     LICENSE      VERSION      composer.json  docs          package.json       rector.php    tests         vendor
CONTRIBUTING.md        Procfile     assets       composer.lock  ecs.php       phpstan.dist.neon  src           translations  webpack.config.js
Dockerfile             README.md    bin          config         migrations    phpunit.xml.dist   symfony.lock  uploads       yarn.lock
Dockerfile-frankenphp  SECURITY.md  codecov.yml  crowdin.yml    node_modules  public             templates     var
root@PartDB:/var/www/partdb# 
```

Tropaion, 2025-09-01:
Yes, thanks, works now.

closed 2025-09-01 by Tropaion

virtadpt, 2025-09-04:
I ran into this problem as well under the same circumstances.  What I wound up doing was re-cloning the Part-DB-server (because the one I'd been updating from was two or three years old), copying the .env.local file into the new clone, and starting over with a brand new build.