Part-DB/Part-DB-server #1013: Allow System Configuration to specify TRUSTED_PROXIES (2.0.2)

bson, 2025-09-01:
Add an item to the System Configuration pages/tabs that permit specifying a trusted proxy list.  Retain it in the DB so it doesn't get wiped on a docker image update.

bson, 2025-09-01:
There's no docker compose step when downloading an image and running it in a container on a Synology.  It needs the reverse ingress proxy to be whitelisted, or external OAuth integration fails with strange errors that are very difficult to deduce.

I also don't understand what kind of deployment scenario doesn't include a controlled ingress point, like a load balancingreverse proxy, with a network of servers behind it?  If reverse proxies aren't to be trusted, then this is configured as a policy at the ingress point, not on individual servers behind it.

jbtronics, 2025-09-01:
Still you can set environment variables on Synology docker containers; https://kb.synology.com/en-br/DSM/help/Docker/Docker?version=6

Everything else would be odd, as env variables the typical way to configure docker applications....

If you run Part-DB directly on a Webserver, the Webserver is the point of ingress. No need for a reverse proxy. Setting TRUSTED_PROXIES then would actually be a mild security risk, as every network user could then fake his IP address by sending the correct headers, which circumvents IP logging and certain security protects like rate limiting...

There is no way for an application to know if the thing it gets his requests is a reverse proxy or a malicious attacker. That's why you have to white list the trusted proxies IP address, so that Part-DB knows whose x-forwarded headers it can trust and which not. 

But that will depend on your exact situation and application stack so there is no good default value for this...

closed 2025-09-03 by jbtronics