Part-DB/Part-DB-server #1017: Can't search info provider Mouser after upgrade to 2.0.2

w0hc, 2025-09-03:
**Describe the bug**
500 Error when attempting to search info provider Mouser.com. Worked in previous versions <2.0.x
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. New Part
2. Create Part from Info Provider
3. Enter keyword
4. Select Provider Mouser
5. Search

Doing the same with DigiKey still works fine.

**Logs**
`[2025-09-02T23:43:21.111678+00:00] request.CRITICAL: Uncaught PHP Exception RuntimeException: "Unknown response format: {"Errors":[{"Id":0,"Code":"Invalid","Message":"Invalid unique identifier.","ResourceKey":"InvalidIdentifier","ResourceFormatString":null,"ResourceFormatString2":null,"PropertyName":"API Key"}],"SearchResults":null}" at MouserProvider.php line 213 {"exception":"[object] (RuntimeException(code: 0): Unknown response format: {\"Errors\":[{\"Id\":0,\"Code\":\"Invalid\",\"Message\":\"Invalid unique identifier.\",\"ResourceKey\":\"InvalidIdentifier\",\"ResourceFormatString\":null,\"ResourceFormatString2\":null,\"PropertyName\":\"API Key\"}],\"SearchResults\":null} at /var/www/html/src/Services/InfoProviderSystem/Providers/MouserProvider.php:213)
[stacktrace]
#0 /var/www/html/src/Services/InfoProviderSystem/Providers/MouserProvider.php(135): App\\Services\\InfoProviderSystem\\Providers\\MouserProvider->responseToDTOArray()
#1 /var/www/html/src/Services/InfoProviderSystem/PartInfoRetriever.php(90): App\\Services\\InfoProviderSystem\\Providers\\MouserProvider->searchByKeyword()
#2 /var/www/html/vendor/symfony/cache/LockRegistry.php(111): App\\Services\\InfoProviderSystem\\PartInfoRetriever->{closure:App\\Services\\InfoProviderSystem\\PartInfoRetriever::searchInProvider():86}()
#3 /var/www/html/vendor/symfony/cache/Traits/ContractsTrait.php(102): Symfony\\Component\\Cache\\LockRegistry::compute()
#4 /var/www/html/vendor/symfony/cache-contracts/CacheTrait.php(64): Symfony\\Component\\Cache\\Adapter\\AbstractAdapter->{closure:Symfony\\Component\\Cache\\Traits\\ContractsTrait::doGet():85}()
#5 /var/www/html/vendor/symfony/cache/Traits/ContractsTrait.php(85): Symfony\\Component\\Cache\\Adapter\\AbstractAdapter->contractsGet()
#6 /var/www/html/vendor/symfony/cache-contracts/CacheTrait.php(30): Symfony\\Component\\Cache\\Adapter\\AbstractAdapter->doGet()
#7 /var/www/html/src/Services/InfoProviderSystem/PartInfoRetriever.php(86): Symfony\\Component\\Cache\\Adapter\\AbstractAdapter->get()
#8 /var/www/html/src/Services/InfoProviderSystem/PartInfoRetriever.php(72): App\\Services\\InfoProviderSystem\\PartInfoRetriever->searchInProvider()
#9 /var/www/html/src/Controller/InfoProviderController.php(154): App\\Services\\InfoProviderSystem\\PartInfoRetriever->searchByKeyword()
#10 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(183): App\\Controller\\InfoProviderController->search()
#11 /var/www/html/vendor/symfony/http-kernel/HttpKernel.php(76): Symfony\\Component\\HttpKernel\\HttpKernel->handleRaw()
#12 /var/www/html/vendor/symfony/http-kernel/Kernel.php(182): Symfony\\Component\\HttpKernel\\HttpKernel->handle()
#13 /var/www/html/vendor/symfony/runtime/Runner/Symfony/HttpKernelRunner.php(35): Symfony\\Component\\HttpKernel\\Kernel->handle()
#14 /var/www/html/vendor/autoload_runtime.php(29): Symfony\\Component\\Runtime\\Runner\\Symfony\\HttpKernelRunner->run()
#15 /var/www/html/public/index.php(5): require_once('...')
#16 {main}
"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.1.100","http_method":"POST","server":"part-db.domain.com","referrer":"http://part-db.domain.com/en/tools/info_providers/search"}
[2025-09-02T23:43:21.113351+00:00] security.DEBUG: Stored the security token in the session. {"key":"_security_main"} {"token":{"authenticated":true,"roles":["ROLE_USER"],"user_identifier":"admin"},"url":"/en/tools/info_providers/search","ip":"192.168.1.100","http_method":"POST","server":"part-db.domain.com","referrer":"http://part-db.domain.com/en/tools/info_providers/search"}
192.168.1.100 - - [02/Sep/2025:18:43:20 -0500] "POST /en/tools/info_providers/search HTTP/1.1" 500 4498 "http://part-db.domain.com/en/tools/info_providers/search" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:142.0) Gecko/20100101 Firefox/142.0"
- -  02/Sep/2025:18:43:20 -0500 "POST /index.php" 500`

**Server Side**
- Part-DB Version: 2.0.2
- PHP Version: 8.4.11
- Database Server SQLite
- Environment: Docker

Cleared cache but was of no help.

d-buchmann, 2025-09-03:
Hi,

The runtime error is because the mouser response doesn't contain any search results.
Instead, it complains that your API Key is invalid.
Did you transfer the key from env vars to the settings system using the migration console command?
Have you checked [directly](https://api.mouser.com/api/docs/ui/index#/SearchApi/SearchApi_SearchByKeyword) if your api key is actually still valid?

PS: Please enclose your logs in triple backticks to make them more readable and avoid falsely linking old issues, like so:
\`\`\`
code
\`\`\`

w0hc, 2025-09-03:
Thanks for the reply. I did transfer the key from env vars to the system settings. 
Checking directly gives me this output:
```{
  "Errors": [
    {
      "Id": 0,
      "Code": "Invalid",
      "Message": "Invalid unique identifier.",
      "ResourceKey": "InvalidIdentifier",
      "ResourceFormatString": null,
      "ResourceFormatString2": null,
      "PropertyName": "API Key"
    }
  ],
  "SearchResults": null
}
```

I'll request a new API key and try again once approved.

Thanks for the tip on the triple backticks :)

closed 2025-09-07 by w0hc